RobotLocomotion/drake #19600: Partition optitrack support out of Drake

jwnimmer-tri, 2023-06-14:
Supporting the optitrack driver & messages in the pre-packaged Drake binaries is a bit of a chore.  I'd like to propose splitting it into a completely separate package to smooth things out:

(1) Enhance https://github.com/RobotLocomotion/optitrack-driver with releases that are easy to use.  (For example, provide releases with installable packages that incorporate the generated message types, instead of just source archives.)

(2) Deprecate stuff inside Drake for eventual removal:
- optitrack_client (the program)
- optitrack lcmtypes
- OptitrackSender
- OptitrackReceiver

(3) Adjust Anzu to directly use the driver repo and its message types, instead of via Drake.

\CC @sammy-tri curious to get your feedback.

sammy-tri, 2023-06-15:
If I were just answering for my own use cases (by which I mean any Anzu user I suppose):    (1) not really necessary (2) copy the `LeafSystem`s into anzu, get the types and client from `optitrack-driver`, (3) yes.

To support a broader user community (1) seems necessary but I'm not sure what it looks like (I assume you'd want these packages to be usable by people who've installed the drake binary releases?), (2) where do you package the `LeafSystem`s?

I'm not sure how broad of a user community there is for drake+optitrack.

jwnimmer-tri, 2023-06-15:
Ah, I should have clarified that.  The `optitrack-driver` packages would only have the driver / client / messages.  The leaf systems would be gone for good.  (Users could copy the leaf system code from Drake to their project if they need it.)  I think really nobody uses the sender system, and the receiver system is cake (a few lines of python) and anyway many users would probably would want more application-specific customized versions the receiver it anyway.  In other words, my premise is that LcmSubscriberSystem is sufficient as a "receiver".

sammy-tri, 2023-06-15:
> my premise is that LcmSubscriberSystem is sufficient as a "receiver".

Works for me.

jwnimmer-tri, 2023-07-29:
The `optitrack.client` wheel file has been officially release now, at: https://github.com/RobotLocomotion/optitrack-driver/releases/tag/v0.0.20230714

closed 2023-07-31 by rpoyner-tri

jwnimmer-tri, 2023-09-10:
Yes, the https://github.com/RobotLocomotion/optitrack-driver lives on outside of Drake and provides the program and lcmtypes.  The latest release in that repository has a wheel file with everything, including the generated lcmtypes in all languages.

For Python:

Import the wheel and use its message classes to send and receive messages using Drake's `LcmPublisherSystem` and `LcmSubscriberSystem`.  In many cases, that will be enough -- the systems that need to "speak to optitrack" can operate on the message type as a abstract input/output port.  (The `OptitrackLcmFrameSender` and `OptitrackReceiver` were never bound in pydrake anyway, so that matches the prior status quo.)  If it's still helpful to convert to have systems in Python that convert to some number of of `RigidTransform`s or from a `FrameKinematicsVector`, a user could write a simple Python `LeafSystem` to fit the specifics of what they require.

For C++:

Users who want to use the generated C++ lcmtypes should adjust their build system either to run `lcm-gen` themselves (against the `*.lcm` files in this repository), or else add the `optitrack-driver` wheel as an external workspace dependency (it's just a plain zip file, which the build system can unzip and add the header files to the include path).  That's enough to use `LcmPublisherSystem` and `LcmSubscriberSystem`.  C++ users who want the exact logic of the `OptitrackLcmFrameSender` and `OptitrackReceiver` systems can copy that code [from Drake](https://github.com/RobotLocomotion/drake/blob/v1.20.0/systems/sensors/optitrack_sender.h) into their own project.

PR #19866 (2023-07-31): Deprecate all optitrack-related code for removal
The following code will be removed in ~3 months:

- `drake::systems::sensors::OptitrackLcmFrameSender`
- `drake::systems::sensors::OptitrackReceiver`
- The `@optitrack_driver` external
- The installed `optitrack_client` program

Note that this does not add Bazel deprecation markers anywhere (doing so ends up being extra trouble for downstreams).

Closes #19600.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19866)