RobotLocomotion/drake #19617: Python bindings with container return types may be broken

rpoyner-tri, 2023-06-16:
### What happened?

See #19615 for a concise failing test.

The binding in question is trying to return a vector/list of pointers, and ends up trying to copy the uncopyable point-to objects, and throwing.

### Version

See #19615 code.

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

### Relevant log output

```shell
You may be interested in the relevant bit of stack trace from gdb:

(gdb) bt                                                                                                                                                                                                                                                                                                                                                                                                                                 
#0  0x00007fffe02ae4e1 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6                                                                                                                                                                                                                                                                                                                                                   
#1  0x00007fffdad38a9a in pybind11::detail::type_caster_generic::cast (_src=0x555556d90b40, policy=pybind11::return_value_policy::copy, parent=..., tinfo=0x55555634fac0, copy_constructor=0x0,                                                                                                                                                                                                                                          
    move_constructor=0x0, existing_holder=...) at external/pybind11/include/pybind11/detail/type_caster_base.h:714                                                                                                                                                                                                                                                                                                                       
#2  0x00007fffdae43471 in pybind11::detail::type_caster_base<drake::multibody::Body<double> >::cast (src=0x555556d90b40, policy=pybind11::return_value_policy::copy, parent=...)                                                                                                                                                                                                                                                         
    at external/pybind11/include/pybind11/detail/type_caster_base.h:1066                                                                                                                                                                                                                                                                                                                                                                 
#3  0x00007fffdadf6cd1 in pybind11::detail::list_caster<std::vector<drake::multibody::Body<double> const*, std::allocator<drake::multibody::Body<double> const*> >, drake::multibody::Body<double> const*>::cast<std::vector<drake::multibody::Body<double> const*, std::allocator<drake::multibody::Body<double> const*> >&> (src=std::vector of length 2, capacity 2 = {...}, policy=pybind11::return_value_policy::copy,              
    parent=...) at external/pybind11/include/pybind11/stl.h:161                                                                                                                                                                                                                                                                                                                                                                          
#4  0x00007fffdadb9456 in pybind11::cast<std::vector<drake::multibody::Body<double> const*, std::allocator<drake::multibody::Body<double> const*> >&, 0> (value=std::vector of length 2, capacity 2 = {...},                                                                                                                                                                                                                             
    policy=pybind11::return_value_policy::copy, parent=...) at external/pybind11/include/pybind11/cast.h:1059                                                                                                                                                                                                                                                                                                                            
#5  0x00007fffdacc9394 in operator() (__closure=0x555556aa7178, self=..., body=...) at bindings/pydrake/multibody/plant_py.cc:802                   
```
```

rpoyner-tri, 2023-06-16:
FWIW, there is exactly one non-test use of py_keep_alive_iterable in Drake. Apparently it hasn't been particularly well tested.

rpoyner-tri, 2023-06-16:
The problem happens well before we get to the innards of py_keep_alive_iterable. Having to py:;cast(welded_bodies) in the argument list seals our fate. I'm going to suggest that the type signature of py_keep_alive_iterable is wrong, and suggest instead something like this:

```
diff --git a/bindings/pydrake/multibody/plant_py.cc b/bindings/pydrake/multibody/plant_py.cc
index 79588ec89a..c1f344b121 100644
--- a/bindings/pydrake/multibody/plant_py.cc
+++ b/bindings/pydrake/multibody/plant_py.cc
@@ -799,7 +799,7 @@ void DoScalarDependentDefinitions(py::module m, T) {
             [](const Class& self, const Body<T>& body) {
               auto welded_bodies = self.GetBodiesWeldedTo(body);
               return py_keep_alive_iterable<py::list>(
-                  py::cast(welded_bodies), py::cast(&self));
+                  welded_bodies, py::cast(&self));
             },
             py::arg("body"), cls_doc.GetBodiesWeldedTo.doc)
         .def("GetBodiesKinematicallyAffectedBy",
diff --git a/bindings/pydrake/pydrake_pybind.h b/bindings/pydrake/pydrake_pybind.h
index cb98699f9a..4aa4a52018 100644
--- a/bindings/pydrake/pydrake_pybind.h
+++ b/bindings/pydrake/pydrake_pybind.h
@@ -53,12 +53,13 @@ inline py::object py_keep_alive(py::object nurse, py::object patient) {
 /// Use this to manually cast an iterable type (e.g. py::list, py::set). See
 /// pydrake_pybind_test for an example.
 /// N.B. This should *not* be used for `py::dict`.
-template <typename PyType>
-inline PyType py_keep_alive_iterable(PyType nurses, py::object patient) {
-  for (py::handle nurse : nurses) {
-    py_keep_alive(py::reinterpret_borrow<py::object>(nurse), patient);
+template <typename PyType, typename CxxIterable>
+inline PyType py_keep_alive_iterable(CxxIterable nurses, py::object patient) {
+  py::list kas;
+  for (auto& nurse : nurses) {
+    kas.append(py_keep_alive(py::cast(nurse), patient));
   }
-  return nurses;
+  return PyType(kas);
 }

```

There might be a cleverer way of constructing/initializing PyType in a collection-agnostic way, but i suspect the above works well enough for the types we claim to care about.

EricCousineau-TRI, 2023-06-20:
Relates #11046. Will see if I can revisit the testing there (if any) to see if it's still relevant.

PR #19655 (2023-06-26): [bindings] Use vanilla py::cast() for keep alive behavior
Resolves #19617, at least as far as the acute error goes. Using Rico's solution outright.
~Resolves~ Relates #11046 (core issue) - This narrows the use of `py::detail::keep_alive_impl()`

Closes #19615 (superseded)
Relates #19636 (attempt to make a `type_caster<>` for `copyable_unique_ptr<>`)

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19655)
closed 2023-06-26 by EricCousineau-TRI