RobotLocomotion/drake #19629: Comparing symbolic variable should return more meaningful error message

hongkai-dai, 2023-06-19:
Currently when I compare two symbolic variables in the gtest, the behavior is like this
```cc
symbolic::Variable x("x");
symbolic::Variable y("y");
```
If I do
```cc
EXPECT_EQ(x, x);
```
then the test passes.

If I do
```cc
EXPECT_NE(x, y);
```
then I receive the error message
```
unknown file: Failure
C++ exception with description "The following environment does not have an entry for the variable x

" thrown in the test body.
```

There are two things unexpected
1. The symbolic variable comparison works for `EXPECT_EQ` but not `EXPECT_NE`.
2. The error message complains about the symbolic Environment.

Should we make `EXPECT_NE(x, y)` working?

jwnimmer-tri, 2023-06-19:
Google test is using `operator==` and `operator!=` to compare, in other words it is making a `Formula` first and then implicitly calling `formula.Evaluate()` second using an empty environment.

For comparing structural equality of expressions or variables, we have these functions instead:
https://github.com/RobotLocomotion/drake/blob/master/common/test_utilities/symbolic_test_util.h

I support trying to make this more clear, either by having all of the comparison operators (eq, ne, lt, gt, ge, le) either all fail to compile, or else all check structural equality automatically.

hongkai-dai, 2023-06-19:
> I support trying to make this more clear, either by having all of the comparison operators (eq, ne, lt, gt, ge, le) either all fail to compile, or else all check structural equality automatically.

Sorry I don't quite follow this. If we do want to have the comparison operators to fail, how should we do that? We still want to support `x == y` returns a symbolic Formula.

jwnimmer-tri, 2023-06-19:
Googletest uses a helper class to do the comparison:

```c++
class EqHelper {
 public:
  // This templatized version is for the general case.
  template <
      typename T1, typename T2,
      // Disable this overload for cases where one argument is a pointer
      // and the other is the null pointer constant.
      typename std::enable_if<!std::is_integral<T1>::value ||
                              !std::is_pointer<T2>::value>::type* = nullptr>
  static AssertionResult Compare(const char* lhs_expression,
                                 const char* rhs_expression, const T1& lhs,
                                 const T2& rhs) {
    return CmpHelperEQ(lhs_expression, rhs_expression, lhs, rhs);
  }
```

We could patch the class so that the `enable_if` disallows the case there `T1 == T2` is not `bool`.

jwnimmer-tri, 2023-06-20:
I think my general advice here is:

(1) Adjust EXPECT_EQ and EXPECT_NE to fail when used on Expression.

(2) Provide examples / suggestion of a better comparison to use instead.  That could be `ExprEqual` for now, but I could probably come up with a gtest matcher suggestion that does a better job of printing out the lhs and rhs values.

If you like, you can assign this issue to me to fix.