RobotLocomotion/drake #19641: Hydroelastic Contact: Basics images missing on Deepnote

jwnimmer-tri, 2023-06-20:
### What happened?

Open our online hosted and browse [Hydroelastic Contact: Basics](https://deepnote.com/workspace/Drake-0b3b2c53-a7ad-441b-80f8-bf8350752305/project/Tutorials-2b4fc509-aef2-417d-a40d-6071dfed9199/notebook/hydroelastic_contact_basics-1d03814005274062aad03a875aa5bd86).

There are two problems:

(1) Hard linebreaks.  Apparently Jupyter locally and Deepnote use different markdown conventions.  The hard linebreaks in the source Markdown are showing up in Deepnote and disrupting the flow.  In local Jupyter preview, all is fine.

(2) Missing images.  Apparently, Deepnote cannot handle image attachments.

### Version

1.18.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

dockerhub

jwnimmer-tri, 2023-06-20:
See here for how attachments work: https://jupyter-notebook.readthedocs.io/en/stable/examples/Notebook/Working%20With%20Markdown%20Cells.html#Markdown-attachments

I've asked about Deepnote vs attachments here: https://community.deepnote.com/c/ask-anything/importing-attachments-from-ipynb-upload

DamrongGuoy, 2023-06-21:
According to the answer in Deepnote Community, we'll have to store images outside the notebook file. We also discussed it in this Slack [thread.](https://drakedevelopers.slack.com/archives/C029CJ333CZ/p1687307674508099?thread_ts=1687292051.715169&cid=C029CJ333CZ)  Today I'll do these:

- [  ] Copy images from hydroelastic tutorial to hydroelastic user guide.
- [  ] Replace attached imaged in the tutorial with URL to images in the user guide.

DamrongGuoy, 2023-06-21:
#19643 has merged. I double checked by copying the file to my temporary project in Deepnote.  The linebreaks look reasonable like the following picture.  I'm working on moving images from the tutorial into the user guide.

[image]

closed 2023-06-28 by EricCousineau-TRI

jwnimmer-tri, 2023-06-28:
FYI @DamrongGuoy because the changes from v1.18.0 to master on this notebook are only to the Markdown content (not code), it is safe to copy this new notebook file to our official Deepnote Tutorials project, without waiting for v1.19.0 tag.

It would go a bit like the release playbook, except;
- Do not change the Dockerfile.
- Do not use the maintainer notebook to copy the files.

Instead:
- Open the Deepnote Tutorials.
- Drag the new ipynb file (from a checkout of drake master) into the notebooks tab.
- Delete the old copy of the notebook and rename the new copy to its proper name.
- Run the notebook (in order to populate the cells).

DamrongGuoy, 2023-06-28:
Thank you.  I'll copy the new notebook file to our official Deepnote Tutorials project.

DamrongGuoy, 2023-06-28:
I have copied it to our official Deepnote Tutorials project. Thank you!

PR #19660 (2023-06-28): [tutorials] Make Hydroelastic Contact use images from User Guide
- Remove attached images because Deepnote does not support attachments.
- Also fix section markers "###".

Fix #19641

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19660)