RobotLocomotion/drake #19657: Only ever create one Gurobi environment (GRBenv)

RussTedrake, 2023-06-22:
The usability issue:  every call to gurobi on my academic license spits out e.g.
```
--------------------------------------------
Warning: your license will expire in 14 days
--------------------------------------------

Academic license - for non-commercial use only - expires 2023-07-06
Set parameter Username
```
to the console.  I have programs that solve a LOT of gurobi problems, and these messages completely dominate the console output.

Looking into it (e.g. [here](https://groups.google.com/g/gurobi/c/O7Kht25-OIA)) I believe that we should not be creating a GRBenv on every new instantation of the gurobi solver.  The [gurobi docs](https://www.gurobi.com/documentation/9.5/refman/cs_env_start.html) say
>In general, you should aim to create a single Gurobi environment in your program, even if you plan to work with multiple models. Reusing one environment is much more efficient than creating and destroying multiple environments. The one exception is if you are writing a multi-threaded program, since environments are not thread safe. In this case, you will need a separate environment for each of your threads.

I think we should life the GRBenv* that is currently used in GurobiSolver to be `static thread_local`?

cc @hongkai-dai 
@jwnimmer-tri -- I can do the implementation if you endorse the concept and perhaps give me a little advice on the proper approach to test coverage.

jwnimmer-tri, 2023-06-22:
To set the stage, I'll remind everyone that keeping around a `solver = GurobiSolver()` object will retain the license for the lifetime of that `solver` object.  In some cases that's a simple way around this kind of problem, but here I assume you're using the `solvers::Solve()` free function and so it's not practical to call the same `SolverInteface` handle repeatedly.

To date, our answer for that situation has been to explicitly grab the license once at some point during process start-up, e.g., like this:

```c++
void DoMain() {
  // ... my application code ...
}

int main() {
  // Hold the license during this entire process.  No need to pass this around,
  // the GurobiSolver can find it when it needs it.
  auto gurobi_license = GuribiSolver::AcquireLicense();
  DoMain();
  return 0;
}
```

The Python spelling is analogous (https://drake.mit.edu/pydrake/pydrake.solvers.html#pydrake.solvers.GurobiSolver.AcquireLicense).

Is that good enough to meet the need?

For commercial users that have a cap on the number of concurrent licenses, one difference between the current workflow and what you propose is that the environment (license) is only held while solving (in the current code) versus for the remainder of the process's lifetime (in your proposal).  If the program isn't going to be doing any more solving (e.g., if its an idle Jupyter terminal, or it's in a different mode now like setup phase versus simulation phase), it's usually to our advantage to release the license back into the pool, so that others can use it.  Contention for the small number of licenses is usually high.

> The one exception is if you are writing a multi-threaded program, since environments are not thread safe.

I'm not sure that we took this into account when applying the ScopedSingleton pattern to the Gurobi license.  (It's one per process, not one per thread.) On the other hand, I haven't ever seen a crash.  Might be worth digging in further.

RussTedrake, 2023-06-22:
That does work in my notebook. Thanks! (I had forgotten that solution).  The result for me was
```
# Acquire a Gurobi license once (to avoid constantly reaquiring it).
gurobi = GurobiSolver()
if gurobi.available() and gurobi.enabled():
    gurobi_license = gurobi.AcquireLicense()
```

But we should not have to do that every time in every notebook that might possibly use gurobi, when we can handle it easily enough with a singleton (as recommend) at the solver level?

jwnimmer-tri, 2023-06-22:
> ... when we can handle it easily enough with a singleton (as recommended) at the solver level?

That change would defeat the use case of automatically releasing the license when it is unused, for commercial users who only have 1 or 2 seats available on a shared server.

Perhaps there is a way to ask the GRBenv whether license is coming from a shared server, and only auto-release it under that condition.

RussTedrake, 2023-06-22:
Based on a little digging, I don't see a way to get this information reliably from the API.  But it seems like checking the license file for the existence of the string `SERVER` should be pretty reliable.  E.g. the license file would look something like:
```
SERVER server-name HOSTID=any port-number
USE_SERVER
```
wdyt?

jwnimmer-tri, 2023-06-22:
My license file says:
```
TOKENSERVER=some-hostname.tri.global
```

The safest best would be if we could identify some syntax that indicated it's affirmatively a *local* license, i.e., some pattern from your current license file.

That way, in case of a no-match false negative because the server syntax had changed, we would default to *not* holding the license indefinitely, i.e., we fail closed.

We could also ask Gurobi to add an API to ask this, for some future release of Gurobi.

hongkai-dai, 2023-06-22:
On my standalone GUROBI license, it has
```
TYPE=NODE
```
maybe this means a single node license, instead of one runs on the server?

RussTedrake, 2023-06-22:
Another idea: we could offer a second (optional) environment variable, alongside `GRB_LICENSE_FILE`, to opt-in to holding onto the license?  That way the decision about it could be reasoned by the user at the time of them reasoning about the license?

RussTedrake, 2023-06-22:
It seems that *maybe* the string `HOSTNAME` would be consistent with it being tied to your local machine?  e.g.
```
HOSTNAME=your-hostname HOSTID=your-hostid
FEATURE gurobi gurobi ver-num 31-dec-yyyy uncounted HOSTID=your-hostid
```

jwnimmer-tri, 2023-06-22:
> We could offer a second (optional) environment variable, ...

I am OK with that.  It has the downside that it doesn't happen fully automatically, but if our heuristics for checking whether the license has a seat count are too brittle, then it's probably the best we can do until Gurobi offers an API for this.

If you go that route, be sure to cite it from here:
https://drake.mit.edu/doxygen_cxx/group__environment__variables.html

hongkai-dai, 2023-06-22:
Right, the HOSTNAME and HOSTID match my local machine on my standalone license.

BTW, could you upgrade your academic license? I think you can still use the old GUROBI with a new license.

RussTedrake, 2023-06-22:
So perhaps we go ahead with the HOSTNAME approach?

jwnimmer-tri, 2023-06-22:
OK by me.

cohnt, 2023-06-23:
> BTW, could you upgrade your academic license? I think you can still use the old GUROBI with a new license.

I think you should be able to use the old Gurobi with a new license. My license file has the line
```
TYPE=ACADEMIC
VERSION=10
```
and `gurobi_cl --version` outputs
```
Gurobi Optimizer version 9.5.2 build v9.5.2rc0 (linux64)
```

PR #19713 (2023-07-05): GurobiSolver acquires a local license at most once per process
Iff the contents of the file identified in GRB_LICENSE_FILE contain the magic keywork HOSTID, then any license that is obtained will never be destroyed.

Resolves #19657.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19713)
commit 88255314 (2023-07-05): GurobiSolver acquires a local license at most once per process (#19713)

Iff the contents of the file identified in GRB_LICENSE_FILE contains
the magic keyword HOSTID, then any license that is obtained will never
be destroyed.

Resolves #19657.

A new implementation and unit test strategy
closed 2023-07-05 by sherm1