RobotLocomotion/drake #19680: [pydrake] Disable spdlog redirection when using multithreaded C++ code?

EricCousineau-TRI, 2023-06-26:
**Is your feature request related to a problem? Please describe.**
I am writing Python bindings in a downstream project (Anzu) that does work in a multithreaded fashion, and calls `drake::log()->info()`, `->debug()`, `->trace()`, etc., in those separate threads.

Even though this code doesn't explicitly call Python code in those other threads (e.g. not via `std::function<>` or C++ <-> Python polymorphism), it still seemed to deadlock.

If I commented out this line, my multithreaded code seemed to work fine:
https://github.com/RobotLocomotion/drake/blob/b9813a712454580cd6bc0724f8d2506d85412e58/bindings/pydrake/common/module_py.cc#L103

**Describe the solution you'd like**
Perhaps an environment variable, like `PYDRAKE_NO_SPDLOG_REDIRECT`, or a function to call from pydrake that undoes the effects of `MaybeRedirectPythonLogging()` and related methods.

**Describe alternatives you've considered**
Writing a C++ pybind11 module as follows fixes the issue:

**Additional context**
N/A

jwnimmer-tri, 2023-06-26:
If we end up with an environment variable, my vote to best align with [the current names](https://drake.mit.edu/doxygen_cxx/group__environment__variables.html) would be to name it starting with `DRAKE_`.

Perhaps `DRAKE_PYTHON_LOGGING`, where unset or set-to-empty means the status quo and `DRAKE_PYTHON_LOGGING=0` switches `MaybeRedirectPythonLogging()` to be a no-op.

jwnimmer-tri, 2023-06-27:
> Perhaps an environment variable ... or a function to call from pydrake that undoes the effects of MaybeRedirectPythonLogging ...

Either (or both) of those features sound fine to me.  The environment variable is probably the easier one to implement, but the undo function would probably cover more use cases.

jwnimmer-tri, 2023-06-27:
> ... undoes the effects of MaybeRedirectPythonLogging ...

Even before a Drake feature that encapsulates this, I think end-users can accomplish it already:

```c++
// Remove the pylogging_sink (if it exists).
drake::log()->sinks().resize(1);

// Use only the stderr sink.
dynamic_pointer_cast_or_throw<spdlog::sinks::dist_sink_mt>(drake::get_dist_sink())->set_sinks({
  std::make_shared<spdlog::sinks::stderr_sink_mt>()});
```

PR #19685 (2023-06-29): [bindings] Add env var DRAKE_PYTHON_LOGGING to disable redirect
Resolves #19680

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19685)
PR #19703 (2023-07-31): [py common] Add use_native_cpp_logging feature
Closes #19695.  Closes #19680. Relates #19685.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19703)
closed 2023-06-29 by jwnimmer-tri