RobotLocomotion/drake #19709: Use hydro without manual annotation of every single model file

amcastro-tri, 2023-06-30:
The problem is that we have a lot of models decorated with point spheres (and no hydro specification at all). We know these models would simulate better with hydro.
The ycb objects, for instance, would be a great place to start?  currently users have a mix of boxes + tiny sphere collision geometry. 

While we could default geometries to always have a hydroelastic representation, the tiny spheres people add to their models are incompatible with the hydroelastic model (near to or zero volume). We could ask users to remove those spheres, but this only puts the burden on them. Users would probably like to switch back and forth between models (hydro vs. simple geometry + tiny spheres).

Here are some highlights from a long conversation in Slack that I believe capture the main points and challenges discussed:
- Hydro is awesome, everyone should use it a lot more. How do we get them to do that?
- Adding new Drake-specific XML boilerplate (the hydro representation) to every single SDFormat file in existence is a non-starter. The XML syntax is great for users to tune the hydro representation, but it shouldn't be required.
- Instead, we should add a new MbP ContactModel enum value. The current kHydroelasticWithFallback says that a mixed collision pair (hydro-rep + no-hydro-rep) falls back to point contact. The new enum option should prefer hydro-hydro in the mixed case, by inferring a hydro representation. Eventually, the new enum value should become our default.
- The Dynamics team needs to decide what a good inferred representation should look like.
- After all of that, there is still an edge case to consider: 
  - Models that in the past used primitive box collision with point contact often have tiny little spheres at the box corners. Currently, their extremely small volume (~attoliter) causes an exception when hydrofied. One way or another, they should be rounded down to zero (either don't participate in hydro contact, or participate but are a no-op).
- "This is the scale of things I care about." 
  - With a default value with length unit, the issue of tiny spheres seems to be easily resolved. If for hydro I care about things on the scale of 1e-3m, then a sphere with 1e-7m should be ignored when hydrofied unless otherwise noted.

rpoyner-tri, 2023-08-02:
Had a f2f discussion of this problem with @RussTedrake and @joemasterjohn today. Various techniques were discussed:
* adding code to the parser to add hydro defaults
* adding code into scene graph at some level to add hydro defaults
* various forms of plant-level contact mode hackery
* application-level property hacking after the fact

Some current possible obstacles were discussed:
* currently, the semantics of scene graph properties and geometry types make it likely that liberalizing semantics with defaults may break existing simulations
* the current scene graph reification workflow makes post-hoc property hacking not work
* doing parser modifications may require different semantics/implementations/options for different formats (mujoco vs. sdf, say)

Joe has promised to file a bug about his experience trying to change a geometry post-hoc from hydro-compliant to hydro-rigid. More on that later.

jwnimmer-tri, 2023-08-02:
Why would setting `ContactModel::kHydroelasticInferred` need to set geometry properties at all?  Nothing during parsing, or setting up the SceneGraph, or anything in the public-facing model we expose to users should be any different here.  No existing simulations should be affected.  This should all be under-the-hood.

Some speculation about how I assume it should work:

It seems to me like approximately what we want is when `ContactModel::kHydroelasticInferred` is selected by the user, then we want the `hydroelastic::Geometries::MaybeAddGeometry` function to no longer short-circuit when the user didn't specify any hydro properties.  Instead, it should default the properties and then proceed add the hydro shape.  If there were already registered geometries in the scene graph model when the `length_scale` is first communicated to SceneGraph (to enable inferring a resolution hint), then we would re-add them at that point (as-if they had a properties change).  If the bookkeeping of swapping here is too complicated, we can (for now) insist that the SceneGraph constructor is told up-front whether it should infer the hydro properties, so that it doesn't need to re-jigger later.  The `MultibodyPlantConfig` workflow already accommodates that.

Feel free to invite me to a telecon if you ever need software design help.

rpoyner-tri, 2023-08-02:
I didn't intend to launch a design discussion today; it was already in progress when I reached my desk. I figured it was better to write down what happened and get responses than not.

jwnimmer-tri, 2023-08-02:
Certainly I appreciate the update, thanks.  Nipping in the bud any idea that public `GeometryProperties` should be updated will hopefully save us some time on dead-end designs.

rpoyner-tri, 2023-08-07:
I've begun sketching out the implementation suggested above. For those following this issue, I would love suggestions of specific model files, simulations, or scenarios we should be trying with the proposed new contact mode.

jwnimmer-tri, 2023-08-07:
Above we said "The ycb objects, for instance, would be a great place to start" and I think that's still true.  Things like "cracker box" or "sugar box" are good examples of the tiny-spheres problem.   You can grep for where those model are used.  Things like `authoring_multibody_simulation` or `manipulation_station` are likely candidates.  Possibly a good, anchored workflow would be to try to make the tutorial look like what we aim to achieve here, and then iterate on the subsystems to make it happen.

rpoyner-tri, 2023-08-24:
>Nipping in the bud any idea that public GeometryProperties should be updated will hopefully save us some time on dead-end designs.
Evidently from the abandonment of #19956, I haven't really discovered a way to avoid editing public ProximityProperties. I'm sure I'm missing something obvious.

rpoyner-tri, 2023-08-24:
The problem with never updating public ProximityProperties is that it entails a rewrite of all the MbP-level compliant contact code, which expects to call into an inspector (from QueryObject) to interrogate the properties and make decisions.

I suppose we could hack the inspector returned from QueryObject when "magic hydro" mode is active...

rpoyner-tri, 2023-09-19:
At this point, #20059 is a reasonable sketch of an implementation workflow: SceneGraph's `model` maintains whatever details were parsed into it or modified by model-directed API calls. If a SceneGraph configuration flag is set, then **contexts** made for that SceneGraph will be supplemented by hydro properties, so that most geometries will be hydro-compatible.

Now we come to the next question (validation) -- are we satisfying the user need? In particular, the current sketch has very little sophistication in choosing hydro parameters. The same default set are just spammed across every geometry that does not already have hydro settings in the SceneGraph `model`.  A few exceptions are made for things that obviously won't work (surface meshes can only be hydro/rigid, volume meshes *must* have explicit file or SceneGraph `model` hydro/soft type). This meets the minimum criterion of "don't crash because geometries are not annotated".

The user/roboticist/physicist questions are these:
* do we need to be smarter about how we auto-annotate the properties?
* should parameters like dissipation or resolution hint have defaults chosen based on found geometry, or other clues, rather than just one default across the board?

One practical consideration is that the current implementation sketch does not reach into the layers that actually load mesh files, so it knows essentially nothing about size or geometric properties of mesh objects. That could get changed, if there is a good reason to do so.

Also, I will note that the current strategy does not interfere in any way with explicit annotations or API changes made by the user, so there is a "glide path" between "just do magic hydro" and "now I care about specific parameters/geometries".

jwnimmer-tri, 2023-09-20:
> Do we need to be smarter about how we auto-annotate the properties?

After our chat yesterday, I re-read the [User Guide](https://drake.mit.edu/doxygen_cxx/group__hydroelastic__user__guide.html#hug_visualizing).  I'm a little bit more optimistic now that spamming the same constants onto ever geometry might work OK as an onramp.  Still, I think we should consult our dynamics team and their adjacent roboticists to validate the idea.

rpoyner-tri, 2024-05-16:
See #21443 for a patch that adds SceneGraphConfig to a yaml-consuming example program.

rpoyner-tri, 2024-05-21:
@dmcconachie-tri @rcory The last bit for finishing this issue off is something (example, tutorial, something else?) to help bring developers up to speed on this feature. Since you are likely to be early adopters, I'd like to hear what you think would help. Feel free to comment here, or reach me however is convenient.

jwnimmer-tri, 2024-05-22:
From f2f: final remaining task here is to add a mention to some pre-existing tutorial of this feature, maybe with a little 1-line demo in there somehow.  Just to make users aware this exists.

rpoyner-tri, 2024-06-04:
https://github.com/RobotLocomotion/drake/pull/21481 is merged to satisfy the tutorial request. Declaring victory.

closed 2024-06-04 by rpoyner-tri