RobotLocomotion/drake #19723: For ConvexSet, Clarify the Difference Between Empty and Zero-Dimensional

cohnt, 2023-07-05:
This was spawned from a discussion in #19716, and is related to #19717.

Within ConvexSet and its derived classes, the difference in treatment between empty sets and zero-dimensional sets is somewhat unclear. The set {0}=R^0 is the unique vector space of dimension zero. The empty set is not a vector space at all (since a vector space must always include the zero vector), and is often assigned the dimension -1 (although this is just a convention).

The empty set can be considered as a subset of any dimensional vector space, including the zero-dimensional one. (In fact, there are only two zero-dimensional sets: {} and {0}. It's important to treat these objects distinctly -- for example, an empty HPolyhedron still keeps track of its ambient dimension. Further, towards #19717, it will be important to properly handle zero-dimensional nonempty affine subspaces.

Overall, I'm suggesting that we begin treating these two objects as distinct, and modify/update all parts of ConvexSet and its derived classes to account for this. Specifically, I propose the following changes:

- Treat the empty vector [] as being the unique element of {0}, the zero-dimensional vector space.
- Update several default constructors to have more intuitive mathematical behavior
    - **CartesianProduct**: The empty Cartesian product should certainly be zero-dimensional. By convention, it's also nonempty, so it should be the zero-dimensional set {0}. See [wikipedia](https://en.wikipedia.org/wiki/Empty_product#Nullary_Cartesian_product), or just note the fact that we should be able to add an empty product to any chain of products without changing it.
    - **Intersection**: The empty intersection should always be the whole space. We currently construct it as being zero-dimensional, and I'm fine maintaining this behavior, so it should be treated as the zero-dimensional set {0}.
    - **MinkowskiSum**: The singleton {0} (in any dimension) acts as the identity element for Minkowski addition, whereas {} is annihilatory. We currently construct it as being zero-dimensional, and I'm fine maintaining this behavior, so it should be treated as the zero-dimensional set {0}.
    - **VPolytope**: I think the default constructor in this case should be zero vertices in the zero dimensional space, so {}.
    - **Spectrahedron**: This falls back to the default constructor for ConvexSet (which isn't used anywhere else), creating a zero dimensional set, but not defining whether or not it's empty. Since the documentation suggests it is empty, I suggest we make both Spectrahedron and ConvexSet's default constructor empty by default.
- The other constructors can all remain unchanged.
- Update IntersectsWith to return true if both sets are {0}.
- Update PointInSet to return true when checking if 0 (represented as the empty vector []) is contained in {0}. The tolerance argument will be ignored.
- Continue throwing errors in AddPointInSetConstraints and AddPointInNonnegativeScalingConstraints when the ambient dimension is zero, since the only way this would make sense is if an empty list of variables is passed in, which seems to be clearly bad behavior.

Tagging @RussTedrake, @hongkai-dai, and @AlexandreAmice for further discussion before I start working on a PR.

cohnt, 2023-07-05:
I think we should also disallow the programmer from adding zero-dimensional sets into an CartesianProduct, Intersection, and MinkowskiSum, as in these cases, it either doesn't change the set at all, or trivially makes it empty.

hongkai-dai, 2023-07-06:
> The set {0}=R^0 is the unique vector space of dimension zero.

I agree on the math that the set of zero-dimensional vector has a unique element {0} = R^0, and this set is never empty.
I wonder how to represent {0} = R^0 in the code. Should it be Eigen::VectorXd(0) (equivalently, np.empty((0,)))? How about Eigen::MatrixXd(2, 0) (A matrix with 2 rows and 0 columns), or np.empty((2, 0))?

jwnimmer-tri, 2023-07-06:
> I agree on the math that the set of zero-dimensional vector has a unique element {0} = R^0, and this set is never empty.

I'm no math whiz, but that _seems_ to be at odds with this comment from Russ: https://github.com/RobotLocomotion/drake/pull/19279#pullrequestreview-1406745625.  It would help me out if you could help relate/contrast the current issue's discussion to what was discussed in that review.

hongkai-dai, 2023-07-06:
The comment from Russ is
> I guess if the ambient space is ∅, then the cardinality of (any set in) that space is zero (not 1). So I'd say nullopt.

Which I think is correct. The cardinality of a set is the number of element in the set. So if I have a set of 2D points, such as {(1, 2)}, the cardinality of this set is 1 (because it only has one point), the dimension of the space in which this set lives in is 2, because each element in this set is a 2D vector.

cohnt, 2023-07-06:
> I agree on the math that the set of zero-dimensional vector has a unique element {0} = R^0, and this set is never empty.
> I wonder how to represent {0} = R^0 in the code. Should it be [...]

I think that {0} should be represented by `Eigen::VectorXd(0)` in C++ and `np.empty(0)` in Python. That way, we can always get the zero element in an n-dimensional space with `Eigen::VectorXd::Zeros(n)` or `np.zeros(n)`.

> I'm no math whiz, but that seems to be at odds with this comment from Russ: https://github.com/RobotLocomotion/drake/pull/19279#pullrequestreview-1406745625. It would help me out if you could help relate/contrast the current issue's discussion to what was discussed in that review.

R^0 is a valid ambient space, albeit not a very interesting one (the only possible element is 0). It doesn't make sense to treat ∅ as an ambient space, since it's not a vector space. (By convention, it's usually considered to have dimension -1.) I'd suggest the following behavior for MaybeGetPoint:
- If the ambient dimension is zero, and the set is nonempty, return `Eigen::Vectord(0)`.
- If the ambient dimension is zero, and the set is empty, return nullopt.

jwnimmer-tri, 2023-07-06:
Ah, I finally understand, thanks.  I hadn't considered the concept that when the ambient dimension is zero, we still have two possible sets.

closed 2023-07-20 by sherm1