RobotLocomotion/drake #19734: Release process: publish a stable source archive

jwnimmer-tri, 2023-07-06:
GitHub's auto-generates "Source code (zip)" and "Source code (tar.gz)" attachments for every release we make.  However, these files are _not_ stable.  They will always contain the same source code files after unpacking, but the details of how the archive itself is created will vary, leading to differing archive-file checksums over time.

To resolve this, as part of https://drake.mit.edu/release_playbook.html, we should attach our source tgz archive to the release page.

We have some code in `push_release` script that does this (and pushes it to S3), but that's the wrong stage in the release.  Instead, `download_release_candidate` should create the source archive (and its checksum file), so it gets uploaded at the same time as the binaries.

jwnimmer-tri, 2025-04-19:
> Instead, download_release_candidate should create the source archive (and its checksum file), so it gets uploaded at the same time as the binaries.

Hmm, this might be difficult, since ideally our source release attachment would match GitHub's implicit one, at least on day one.  (GitHub's might change later.)  So we can't upload the source code archive until _after_ placing the tag.

Instead, let's do this as part of the “Post-Release Actions” template, by adding a new checkbox there.

I am not sure exactly how to document or automate it, but the end goal is that "Source code (tar.gz)" link on the release page (which downloads `v1.40.0.tar.gz`) ends up being added back as a new "Asset" attached to the release, named `drake-1.40.0.tar.gz`, and also with the sha256 and sha512 checksum files (so 3 new files uploaded in total).

I do think we could probably have `download_release_candidate` take on this job (or make a new program, if that's easier).  The URL to download (GitHub's default source tar.gz) is easy to compute, so we just need to download it, rename it, add checksums, and then either automatically upload (using the `import github3` RPC API) or just ask the user to attach it by hand.

jwnimmer-tri, 2025-04-22:
(1) Yup.  I would say maybe not in the releases page (we try to keep that short), but we could mention it in https://drake.mit.edu/from_source.html and in the DEE source examples.

(2) Yes.  I want there to be exactly one source of truth and one checksum for what a given release's source code is.

tyler-yankee, 2025-04-22:
Perhaps this PR can also address the outstanding TODO:

https://github.com/RobotLocomotion/drake/blob/17f37909acc2c75d19251568af986df9f37a12d4/tools/release_engineering/download_release_candidate.py#L23-L24

jwnimmer-tri, 2025-04-22:
Fine by me.

tyler-yankee, 2025-04-23:
On naming convention: currently the source archive we push to S3 is `drake-x.y.z.src.tar.gz`:
https://github.com/RobotLocomotion/drake/blob/df780ee6679403b92dc15e8d7a541b8200b13658/tools/release_engineering/dev/push_release.py#L344
Fair to keep that name consistent for the archive we upload to GitHub (rather than `drake-x.y.z.tar.gz` suggested above)?

jwnimmer-tri, 2025-04-23:
Sounds good.

I guess as you found in that code already, should also ensure (by code and/or playbook) that the new release attachment and the file on S3 are identical.

tyler-yankee, 2025-06-23:
With the two above PRs merged, I'm going to close this for now, but it may be reopened if any issues arise when testing on v1.43.0 next month.

PR #22923 (2025-05-20): [tools] Push an official source code archive to GitHub releases
Closes #19734. Drake's release process currently relies on GitHub's automated source archival, for which the creation and checksums aren't necessarily stable over time. This PR introduces an additional step in the post-release process to download GitHub's source code archive, calculate sha256 and sha512 checksums, and re-upload those three files as `drake-<version>-src.tar.gz(.sha256|.sha512)`.

* Factors out some common logic from `push_archive`
* Adds a push of the source code archive to GitHub using the `github3` API in `_push_ [... cut]