RobotLocomotion/drake #19735: Use better deb names in the GitHub release attachment

jwnimmer-tri, 2023-07-06:
@BetsyMcPhail I don't remember what we were thinking about what I wrote this TODO down in https://github.com/RobotLocomotion/drake/issues/18659#issuecomment-1613675809.  Do you recall?  If so, could you elaborate here?

BetsyMcPhail, 2023-07-07:
I think it was to try to avoid the renaming of .deb files in the `push_release` script. @mwoehlke-kitware do you remember the details?

jwnimmer-tri, 2023-07-07:
I think the attachment names need to be unique, so I don't think we can remove "focal" or "jammy" from them.

Possibly the bug is that `push_release` is renaming them into per-distro subdirectories.  I don't see why we need to do that.  The S3 release folder should be a simple backup copy of the GitHub release artifacts.

How about:

(1) `push_release.py --deb` just copies the unrenamed files to the same place as all of the other release artifacts, and 

(2) The apt-site downloader command `curl --fail --location --remote-name "https://drake-packages.csail.mit.edu/drake/release/${platform}/${filename}"` changes to something more like `curl --fail --location --remote-name "https://drake-packages.csail.mit.edu/drake/release/drake-dev_${source_version}-1_amd64-${platform}.deb""` instead.

Would that work?

mwoehlke-kitware, 2023-07-10:
I wonder if you wrote "deb" where you actually meant "binary tarball[s]". Shouldn't there be a bullet to use the actual release version rather than the build date for the latter?

jwnimmer-tri, 2025-01-13:
> Shouldn't there be a bullet to use the actual release version rather than the build date for the latter?

That's already been fixed.

BetsyMcPhail, 2025-01-27:
Change has been merged in. Per f2f discussion, we will carefully check the release script next time it is run and handle any issues if they arise.

closed 2025-01-27 by BetsyMcPhail