RobotLocomotion/drake #19759: DependencyTracker::NotifySubscribers is extremely expensive

pierricgimmig, 2023-07-12:
### What happened?

Around 10% of our "AdvanceTo" call is spent in deep recursive calls to `DependencyTracker::NotifySubscribers`, which is by far the most expensive function in the Drake simulation for our use case, more so than any of the actual core math/physics functions in the simulation.

[image]

[image]

Is there a way to make this faster?

### Version

1.17.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

sherm1, 2023-07-12:
I would note that 90% of your simulation time is _not_ dependency notification so even if the cost were 0 you'd run only 10% faster. That said, I do think we can reduce that substantially -- #9261 laid out some possible speedups that we haven't pursued. Might be time!

BTW it would be helpful to have a relevant benchmark. Can you provide yours or describe the structure? It's unusual to see such a long chain of dependencies.

jwnimmer-tri, 2023-07-12:
FWIW, sometimes I've seen too much invalidation happen when System(s) fail to declare the dependency tickets of their ports, e.g., if an output port doesn't depend on some or all inputs, or only depends on discrete state but not continuous state, etc., it can help to set `prerequisites_of_calc` to something other than "all sources".

jwnimmer-tri, 2023-07-13:
We won't be able to do anything here unless you can provide a reproduction recipe for this bug.  We're not going to try to guess what you're doing that results in slowness.

pierricgimmig, 2023-07-13:
> We won't be able to do anything here unless you can provide a reproduction recipe for this bug. We're not going to try to guess what you're doing that results in slowness.

I totally understand. I'll try to find a public Drake scene that exhibits the same behavior.

jwnimmer-tri, 2024-03-16:
We just landed #21134, which might happen to help here.  In any case, until we have some kind of outline of the situation that's slow, there's no action we can take here for now.  Closing as "not planned", pending more information.

closed 2024-03-16 by jwnimmer-tri

pierricgimmig, 2024-03-20:
Thanks for the heads up, we'll try it out on our side, and update here. Cheers!