RobotLocomotion/drake #19783: MakeSemidefiniteRelaxation does not handle bounding box constraints

bernhardpg, 2023-07-14:
### What happened?

It seems that the newly added `MakeSemidefiniteRelaxation` function currently ignores bounding box constraints. This is evident from the [source code](https://github.com/RobotLocomotion/drake/blob/f6838bcce2ac7d9be936487ef690e691e87d913b/solvers/semidefinite_relaxation.cc#L29), but can be reproduced with the following minimal example

```python
from pydrake.solvers import MakeSemidefiniteRelaxation, MathematicalProgram

prog = MathematicalProgram()
x = prog.NewContinuousVariables(1, "x")[0]
prog.AddLinearConstraint(x >= 3) # will be added as a bounding box constraint

relaxed_prog = MakeSemidefiniteRelaxation(prog)
print(len(relaxed_prog.GetAllConstraints()))
# len(relaxed_prog.GetAllConstraints()) = 2
# The first constraint is Y[-1,-1] = 1, the second is Y must be PSD.
# i.e. the bounding box constraint is not added
```

The bounding box constraints should be added as both linear constraints [here](https://github.com/RobotLocomotion/drake/blob/f6838bcce2ac7d9be936487ef690e691e87d913b/solvers/semidefinite_relaxation.cc#L111), and included in the "big" A matrix [here](https://github.com/RobotLocomotion/drake/blob/f6838bcce2ac7d9be936487ef690e691e87d913b/solvers/semidefinite_relaxation.cc#L117).

### Version

1.19

### What operating system are you using?

macOS 12 (Monterey)

### What installation option are you using?

pip install drake

hongkai-dai, 2023-07-14:
I think we can iterate through every constraint in the original program, and add the convex constraints to the relaxed program (including bounding box, linear, second order cone, SDP, exponential cone, etc), and through an error if we have generic nonlinear constraints.

RussTedrake, 2023-07-14:
@hongkai-dai -- I'd like to keep this issue to just be about the bounding box constraints.  The rest are currently covered (as not supported) by the [AreRequiredAttributesSupported](https://github.com/RobotLocomotion/drake/blob/4f3390794730efafed3e044c0de41fafccf53d61/solvers/semidefinite_relaxation.cc#L38) lines in the current implementation.  It was just the bounding box constraints that I missed in my latest version of the code.

And I believe @bernhardpg is going to contribute the fix.

RussTedrake, 2023-07-31:
Resolved in #19795.

closed 2023-07-31 by RussTedrake