RobotLocomotion/drake #19809: Mesh warnings from franka_description with recent Drake versions

jwnimmer-tri, 2023-07-19:
### What happened?

In `drake/manipulation/models/franka_description/urdf` we provide some sample URDFs for the Franka Panda robot.

If one of those models is loaded into a scene that contains render engines (e.g., for camera simulation) then there are a log of warnings spammed to the console:

For example:
```
[console] [warning] warning: Drake currently only supports OBJs that use a single material across the whole mesh; for drake_models/franka_description/meshes/visual/hand.obj, 5 materials were used: 'Part__Feature001_008_005', 'Part__Feature002_005_005', 'Part__Feature005_001_005', 'Part__Feature005_001_005_001', 'Part__Feature_009_005'. The parsed materials will not be used.
```

It's not OK for Drake models to trigger Drake warnings.  We either need to fix the model, or nerf the warning.

For now, our plan is to nerf the warning.

### Version

1.19.0

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

closed 2023-07-20 by sammy-tri

RussTedrake, 2023-07-23:
I just came to this after hitting a similar wall of warnings in the MIT Quest project (using Drake 1.19).  Nerf'ing the warning is a fine fix for now.  But I wanted to push back on one of the statements above -- we should be fixing the parser (to support multiple materials), not fixing the few models we curate in Drake to conform to a deficient parser.  Everyone else will hit the errors on their models, too.

jwnimmer-tri, 2023-07-27:
I'm not sure what specific statement you are pushing back on?  Nothing in this ticket should be taken to read that supporting multiple materials would be undesirable, nor do I think I've even implied that anywhere else.

The franka models were somewhat unique.  They used multiple diffuse materials, but with no texturing.  Once we have textures involved, things get a lots easier (we can just "bake" the texture with all the materials in one place).  In any case, the goal of this ticket was met -- our pre-installed models were spammy; now, they are not.

PR #19815 (2023-07-20): [geometry] Demote warning message for too many materials
Closes #19809.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19815)