RobotLocomotion/drake #19811: Add visualization of triads to Meldis

AdityaBhat-TRI, 2023-07-19:
**Is your feature request related to a problem? Please describe.**
As of now, it's not easy to visualize frames in Meldis. Restoring this feature (which was once a part of drake visualizer) is important now as it becomes very difficult to visualize the frame transforms in simulation.  

**Describe the solution you'd like**
Restore the feature of visualization of frames (similar to the drake visualizer).

**Describe alternatives you've considered**
-

**Additional context**
-

jwnimmer-tri, 2023-07-19:
Adding some cross references / search keywords:
- Slack thread: https://app.slack.com/client/T0JNB2DS4/C43KX47A9/thread/C43KX47A9-1686755592.829789
- LCM channel name: `DRAKE_DRAW_FRAMES*`.

AdityaBhat-TRI, 2023-07-25:
PR in https://reviewable.io/reviews/RobotLocomotion/drake/19848

jwnimmer-tri, 2023-07-31:
Fixed in #19848.

closed 2023-07-31 by jwnimmer-tri