RobotLocomotion/drake #19820: MeshcatVisualizer illustration alpha fights objects with materials

SeanCurtis-TRI, 2023-07-20:
### What happened?

When the geometries specified have intrinsic material definitions, the illustration alpha slider doesn't work. Observe the following examples:

## Wavefront OBJ with .mtl

Changing the illustration alpha can change the mesh in two ways:

1. The underlying mesh color (defined as `Kd r g b ` in the .mtl file) is *replaced* by the visualizer's default color (typically 90% grey). This is a permanent change.
2. The object may or may not actually become transparent:
    a. If the .mtl file explicitly includes a line that says `d = 0.####` (where the value is decidedly *not* 1), then the mesh will become transparent, but its transparency value is slaved to the slider value and not the original transparency.
   b. If the .mtl file doesn't include the `d = ...` line, or it says `d = 1.0`, the alpha slider won't change its transparency at all.

### Example: .obj with diffuse color -- object has transparency (`d = 0.80000`)

`bazel run //tools:model_visupackage://drake/examples/hydroelastic/ball_plate/plate_8in_col.obj`

[image]
Initial state -- default alpha = 1.0. The plate already has 80% opacity. It has been specifically configured to be partially transparent.

[image]
Bump alpha slightly. The plate's opacity changes by *more* than 2%; it jumps up to the slider value. Finally the color changes *radically*.

Reducing the alpha further reduces the alpha value of the plate. However, its color has been permanently changed and its transparency is now mirrors the slider value.

Modify the opacity value (change the line `d = 0.80000` to `d = 1` in `plate_8in_col.mtl`), to observe the behavior where the slider doesn't affect it.

### Example: textured .obj with no transparency

`bazel run //tools:model_visualizer -- package://drake/manipulation/models/ycb/sdf/006_mustard_bottle.sdf`

[image]
Initial state -- default alpha = 1.0.

[image]
Bump alpha slightly; it doesn't become transparent, but it does become 10% *darker*. It's hard to see, but, trust me, it happens.

[image]
Move alpha to zero; the appearance remains unchanged. Restoring alpha to 1 does *not* restore the mesh to its original brightness, but leaves it at the 90% brightness level.

## glTF

Like with .obj, a .gltf file will have its underlying diffuse color stomped on by the visualizer's default color. There is no appreciable difference between a glTF file that has a diffuse texture vs a diffuse color.

However, its behavior with respect to transparency is different and appears to be influenced by three different parameters:
  - double-sided-ness
  - `alphaMode`
  - Diffuse opacity

Reproduction is problematic as Drake does not currently have many .gltf files accessible to model_visualizer. But we'll try.

### Baseline

The .gltf file is double sided, with "opaque" alpha mode, and has no diffuse opacity.

`bazel run //tools:model_visualizer -- package://drake/examples/scene_graph/sun.sdf`

Behavior
 - Bumping the illustration alpha darkens the sun (as white becomes light grey).
 - As illustration alpha goes to zero, the sun becomes transparent (and opaque as alpha goes to one). This is good.

### AlphaMode = "BLEND"

Apply the following patch (changing the `alphaMode` property of the sun to "BLEND") and re-run visualizer.

```diff
--- a/examples/scene_graph/sun.gltf
+++ b/examples/scene_graph/sun.gltf
@@ -27,7 +27,8 @@
                                        "index":0
                                },
                                "metallicFactor":0
-                       }
+                       },
+                       "alphaMode": "BLEND"
                }
        ],
        "meshes":[
```

Behavior
  - Same as the baseline with one exception, when illustration alpha goes from one to any other value and then *back* to 1, the sun exhibits rendering problems (in some sense, turning itself inside out).

### Double-sided-ness

If the mesh isn't marked as double sided, it's transparency won't change at all. Undo your changes to sun.gltf and apply the following change (and run it again):

```diff
--- a/examples/scene_graph/sun.gltf
+++ b/examples/scene_graph/sun.gltf
@@ -20,7 +20,7 @@
        ],
        "materials":[
                {
-                       "doubleSided":true,
+                       "doubleSided":false,
                        "name":"sun_mat",
                        "pbrMetallicRoughness":{
                                "baseColorTexture":{
```
Behavior:
   - Sliding illustration alpha dims the sun, but it never becomes transparent.

### Diffuse transparency

If the material's inherent transparency is anything other than opaque, its "native" transparency is replaced by the slider value (and also suffers from the same "inside-out-problem" because `"alphaMode"="BLEND"` is a pre-condition of a semi-transparent object). Again, revert changes to sun.gltf and apply the following patch:

```diff
--- a/examples/scene_graph/sun.gltf
+++ b/examples/scene_graph/sun.gltf
@@ -23,11 +23,18 @@
                        "doubleSided":true,
                        "name":"sun_mat",
                        "pbrMetallicRoughness":{
+                               "baseColorFactor":[
+                                       1,
+                                       1,
+                                       1,
+                                       0.5
+                               ],
                                "baseColorTexture":{
                                        "index":0
                                },
                                "metallicFactor":0
-                       }
+                       },
+                       "alphaMode":"BLEND"
                }
        ],
        "meshes":[
```

# Analysis

One contributing problem comes from how `MeshVisualizer` [remembers an objects alpha value](https://github.com/RobotLocomotion/drake/blob/a89156497f0bb225ab9dc05b933b28cbc4a89e1f/geometry/meshcat_visualizer.cc#L206-L207). For objects with materials, it is considered a defect to also specify a `(phong, diffuse)` property. This is why as the slider acts, it introduces the visualizer's default color. We don't actually know the color and alpha of the mesh based on its material, so we get the wrong baseline color *and* the wrong alpha.

The remaining artifacts are far more nuanced and have to do with how Meshcat and three.js deal with object materials and these properties. It is not clear to me which properties need to be considered generally; that requires further investigation.

SeanCurtis-TRI, 2023-07-20:
cc @jwnimmer-tri This is what I was alluding to. The illustration alpha slider is problematic and is non-trivial to solve.

jwnimmer-tri, 2023-07-31:
Idea from group chat with Rico -- maybe the alpha sliders should set a _different_ geometry property (i.e., not "phong,diffuse").  That might make it easier to properly merge / ignore the value.  (It also probably obviates the need to cache the "prior color" / "original color" mapping like we do in some places.)

PR #20140 (2023-09-11): [vis] MeshcatVisualizer and Meldis rely on meshcat APIs to handle alpha sliders
Previously, meshcat visualizer was trying to modulate the opacity of geometries by storing the underlying color/opacity and subsequently setting it by setting the "color" property. This works for primtive shapes but breaks down for complex geometries (e.g., obj with mtl files and glTF files).
    
Instead, MeshcatVisualizer makes use of the new affordances in meshcat. With meshcat, one can set the "opacity" or "modulated_opacity" property. As with setting the "color" property, meshcat will update all of the materia [... cut]
closed 2023-09-11 by SeanCurtis-TRI