RobotLocomotion/drake #19822: Need a developer workflow to generate `drake_repository_metadata.json` for `RobotLocomotion/models`

RussTedrake, 2023-07-20:
### What happened?

I'm trying to add models to Robotlocomotion/models following the instructions in `model_version_control.md`, but I think perhaps the steps have changed?  Simply using local_repository_overload gives me the error:
```
ERROR: /Users/russt/drake/multibody/parsing/BUILD.bazel:59:8: no such target '@drake_models//:drake_repository_metadata.json': target 'drake_repository_metadata.json' not declared in package '' defined by /private/var/tmp/_bazel_russt/9c100c86c1cc4c543d01c45f7a63d5c6/external/drake_models/BUILD.bazel (Tip: use `query "@drake_models//:*"` to see all the targets in that package) and referenced by '//multibody/parsing:drake_models_json'
```

@jwnimmer-tri confirmed that "It's generated from `tools/workspace/github.bzl`.  I'll need to do a real fix, but for now I think you could try remove the data citation in `BUILD`, and nerf `GetDrakeModelsRemoteParams()` in the code to return a default-constructed object instead of using the json.

jwnimmer-tri, 2023-07-20:
The other work-around is to avoid `local_repository_override` and instead change `repository = "RobotLocomotion/models"` to be `repository = "MyUser/models"`, laundering the local changes through push/pull to GitHub.

RussTedrake, 2023-07-20:
Good to know that would work.  I already considered that step 2, and could use that for now.

RussTedrake, 2023-07-20:
Reducing priority, since the work-around mean it is not an immediate blocker.

RussTedrake, 2023-07-23:
Related: Looks like I must have missed a step in https://github.com/RobotLocomotion/drake/pull/19771; building the python bindings now warns about
```
DEBUG: /home/russt/drake/tools/install/install.bzl:75:14: @//examples/manipulation_station:install_data installing file external/drake_models/manipulation_station/bin.mtl which is not in current package
DEBUG: /home/russt/drake/tools/install/install.bzl:75:14: @//examples/manipulation_station:install_data installing file external/drake_models/manipulation_station/bin.obj which is not in current package
DEBUG: /home/russt/drake/tools/install/install.bzl:75:14: @//examples/manipulation_station:install_data installing file external/drake_models/manipulation_station/bin.png which is not in current package
DEBUG: /home/russt/drake/tools/install/install.bzl:75:14: @//examples/manipulation_station:install_data installing file external/drake_models/manipulation_station/table_wide.mtl which is not in current package
DEBUG: /home/russt/drake/tools/install/install.bzl:75:14: @//examples/manipulation_station:install_data installing file external/drake_models/manipulation_station/table_wide.obj which is not in current package
DEBUG: /home/russt/drake/tools/install/install.bzl:75:14: @//examples/manipulation_station:install_data installing file external/drake_models/manipulation_station/table_wide.png which is not in current package
DEBUG: /home/russt/drake/tools/install/install.bzl:75:14: @//examples/manipulation_station:install_data installing file external/drake_models/manipulation_station/shelves.mtl which is not in current package
DEBUG: /home/russt/drake/tools/install/install.bzl:75:14: @//examples/manipulation_station:install_data installing file external/drake_models/manipulation_station/shelves.obj which is not in current package
DEBUG: /home/russt/drake/tools/install/install.bzl:75:14: @//examples/manipulation_station:install_data installing file external/drake_models/manipulation_station/shelves.png which is not in current package
```
But I can't see what step I might have missed...

I will update the issue title with the real victory condition: an updated `model_version_control.md` that actually works.

PR #19876 (2023-08-01): [doc] Fix model_version_control instructions
Closes #19822.

+@rpoyner-tri for feature review, please.
+@RussTedrake for platform review, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/19876)
closed 2023-08-01 by rpoyner-tri