RobotLocomotion/drake #19823: Need a mechanism to pre-fetch `drake_models` before running unittests.

RussTedrake, 2023-07-20:
### What happened?

My pip CI for the manipulation repo [appears to be downloading the models repo on each test](https://github.com/RussTedrake/manipulation/actions/runs/5596599148/jobs/10233839340)... I thought I had avoided that by running
```
          PYTHONPATH=/opt/drake/lib/python3.10/site-packages python3 -c 'from pydrake.all import PackageMap; PackageMap().GetPath("drake_models")'
```
once in the post-checkout setup script.  

@jwnimmer-tri confirmed that:

The problem is in this code:
```
  // Find and/or create the cache_dir.
  internal::PathOrError try_cache = internal::FindOrCreateCache("package_map");
  if (!try_cache.error.empty()) {
    throw std::runtime_error(fmt::format(
        "PackageMap: when downloading '{}', could not create temporary cache "
        "directory: {}",
        package_name, try_cache.error));
  }
  const fs::path cache_dir = std::move(try_cache.abspath);

  // See if the package has already been fetched.
  const fs::path package_dir = cache_dir / GetCacheRelativePath();
  std::error_code ec;
  if (fs::is_directory(package_dir, ec)) {
    mutable_path->set_fetched_path(package_dir.string());
    return path_.get_without_fetching();
  }
```
The code assumes that the cache_dir should be used for both reading (to see if the file is already there) and writing (to download the file in case its needed).  For unit tests, the place we write downloads to is always going to be TEST_TMPDIR, and since we're currently using that for reading also, the pre-downloaded copy in your homedir is never used.
The fix would be something like change FindOrCreateCache to take an argument bool allow_test_tmpdir that can skip the test_tmpdir, and then in the package map code teach both FindOrCreateCache and FindInCache to check if the non-tmpdir cache for the existing file, and if found copy from there it to the tmpdir cache (iff tmpdir was in use). 

Possibly a work-around for now might be to edit your unit test's setUp function to del os.environ["TEST_TMPDIR"] to avoid the two-cache-dirs problem.  That would only work if nothing else in the test needs a tempdir, though.
Maybe better, you could unset that env var immediately before calling PackageMap.GetPath and put restore it immediately after, so:
```
def setUp(self):
  self.package_map = PackageMap()
  tmpdir = os.environ.pop("TEST_TMPDIR")
  self.package_map.GetPath("drake_models")
  os.environ["TEST_TMPDIR"] = tmpdir
  # ... When running tests, use `self.package_map` for parsing.
```

Another work-around: if `drake_models/package.xml` is a bazel data = ... dependency, then it will use that instead of remote packages.  I think if you add drake_models to your workspace, and as a data = [] dependency of your tests, it would be cached (by bazel).

RussTedrake, 2023-07-20:
> Another work-around: if drake_models/package.xml is a bazel data = ... dependency, then it will use that instead of remote packages. I think if you add drake_models to your workspace, and as a data = [] dependency of your tests, it would be cached (by bazel).

To be clear, is this proposal to add `RobotLocomotion/models` via `github_archive` in the `WORKSPACE` of my manipulation repo?  Would that mean forking the `github.bzl`? (It doesn't look like those bzl files are installed?)

RussTedrake, 2023-07-20:
Thank you. I will wait for the true fix unless/until I experience an excruciating number of timeout failures.

jwnimmer-tri, 2023-07-30:
Okay, so _most_ of the manipulation repo CI coverage is using `drake_bazel_installed`, which uses bazel repository caching for the `drake_models` download, and seems to be working as intended (no redundant downloading).

It's only the `jammy-pip` CI configuration that fails to use the bazel caching.  It's quite a weird configuration.  It's using Bazel to run the tests, but its manually setting PYTHONPATH to the installed `/opt` copy of Drake instead of using `drake_bazel_external`.  That's not a configuration that I think Drake should support.  It's going out of the way to do it wrong.

Luckily, however, the failure mode in that case is the same as if someone did `rules_python` + `pip_compile` to pull in `pip install drake`, which _is_ a use case we should support.  The `pip` use case also currently downloads too much, as mentioned [here](https://github.com/RobotLocomotion/drake/pull/19104#pullrequestreview-1369557658):

> ... What still remains to be solved is ... if they are using `rules_python` with `pip` instead ...

So, I'll rename this issue to cover the "Bazel + pip install drake" lack of caching.  Hopefully fixing the `pip install drake` models caching should also solve your CI problem -- or at least, make it a step closer to solving.

RussTedrake, 2023-07-30:
Thank you.  I believe the root cause for my current approach is something like this:
- i often want the manipulation repo to use nightly Drake releases.
- I cannot currently specify the nightly revision in my `setup.cfg` because of #18792, specifically [this](https://github.com/RobotLocomotion/drake/issues/18792#issuecomment-1431994451).

If resolving #18792 or switching to `rules_python` can get me around this, that would be ideal!

jwnimmer-tri, 2023-11-15:
Per f2f, we think Russ has found a work-around.

In any case, I still intent to keep this bug open to track a slightly more general request -- how to help guide Drake's remote package map to use the homedir cache, even when running hermetic bazel tests.  This will help for remote package besides just `drake_models`.