RobotLocomotion/drake #19842: Minimal reproduction of very unstable contact forces for robotiq gripper simulation

RussTedrake, 2023-07-25:
### What happened?

I've carved out a minimal example of the poor performance we've been seeing in the contact modeling with the robotiq gripper.  I thought it was specific to SAP, but i've included two models here: one with SAP + coupler constraints, and another with TAMSI + bushing elements.  Both exhibit massive chattering in contact force directions.  The problem persists even if the timestep is reduced considerably.

The reproduction is in [this notebook](https://www.dropbox.com/scl/fi/dlpdmvpd3r3ibc6uqo2bl/robotiq.ipynb?rlkey=k5ldfx31bkz5xwrl35pbr0w8v&dl=0).  (I can't upload the ipynb directly)

You can view the behavior in [this meshcat recording](https://www.dropbox.com/scl/fi/ic10qvivb681k23xny9q8/robotiq_animation.html?rlkey=5ik0zmo1td7qfxwmv5xbeq8xa&dl=0).

[image]

### Version

master

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

RussTedrake, 2023-07-25:
I've ported a version of the first test to C++ for you here:
https://github.com/RussTedrake/drake/pull/new/robotiq_minimal_repro

It should be trivial to switch in the second model (the last cell of the notebook) if you want to look more closely at that one, too.  I still recommend running this notebook first.

xuchenhan-tri, 2023-10-24:
The link to the c++ repro seems to be wrong. Here is a correction:

https://github.com/RussTedrake/drake/tree/robotiq_minimal_repro

xuchenhan-tri, 2023-10-24:
It seems to me that this is more of a model issue than a solver issue. The collision geometry of the finger tip is messed up. One of the finger tip is modeled as a matrix of tiny spheres for point collision and the other finger tip is modeled as a box, all using point contact. The jittering is likely caused by the well-known point contact issue between two box faces.

After manually curating the asset and replacing all the tiny spheres with hydroelastic, the contact force is a lot more stable:
https://github.com/xuchenhan-tri/drake/tree/robotiq

RussTedrake, 2023-10-25:
I'm sorry that this was still open.  Alejandro and I iterated on it and resolved it similarly (and did even better with hydro)

closed 2023-10-25 by RussTedrake