RobotLocomotion/drake #19860: Copy-paste bot list of all Production builds

jwnimmer-tri, 2023-07-27:
**Is your feature request related to a problem? Please describe.**

For some pull requests, the developer would like to opt-in to certain configurations that are not automatically triggered.

At the moment, figuring out the list of possible 'bot commands is a little difficult.

We give a few samples at https://drake.mit.edu/jenkins.html (e.g., `@drake-jenkins-bot mac-x86-monterey-clang-bazel-experimental-release please`) but not the comprehensive list.

**Describe the solution you'd like**

We should offer some URL where the developer can see the full list of **experimental** build commands that cover the set of **Production** builds.

Here's [a gist](https://gist.github.com/jwnimmer-tri/9b105198b4f9998853506824f0a2b3b7) with the _current_ list of commands we'd want to see.  However, as [drake-jenkins-jobs](https://github.com/RobotLocomotion/drake-jenkins-jobs) evolves we'd want this list to automatically change as well, so we can't just use a plain gist.

For usability, I think possibly we should have two lists -- one with provisioned builds, and one with unprovisioned.  Usually a developer will only want one or the other list.

**Describe alternatives you've considered**

At the moment, this is what I tell developers to do:

(1) Open https://drake-jenkins.csail.mit.edu/view/Production/ -- this is the precise list of all supported build options.
(2) Copy-paste the list of build names from that page into a text editor buffer, with one build name per line.
(3) In the text editor, search-replace the frequency keyword, i.e., `continuous` => `experimental` and `nightly` => `experimental` and `weekly` => `experimental`.
(4) In the text editor, kill any builds you don't want, e.g., search-replace `.*sanitizer.*` => `(empty string)` or `debug` => `release`, etc. (or just use manual editing without regexs).
(5) In the text editor turn the list of builds into bot command, by replacing `(.*)` => `@drake-jenkins-bot \1 please`.
(6) Paste the text editor buffer into a GitHub comment to command the builds.

The goal here is to provide the steps 1,2,3,5 already pre-munged.

svenevs, 2023-07-27:
quick and dirty solution go to https://drake-jenkins.csail.mit.edu/script

```
hudson.model.Hudson.instance.getView('Production').items.each() {
  String experimental_name = it.fullDisplayName.replace("continuous", "experimental").replace("nightly", "experimental");
  org.jenkinsci.plugins.workflow.job.WorkflowJob experimental_job = hudson.model.Hudson.instance.getJob(experimental_name);
  if (experimental_job != null) {
    println "@drake-jenkins-bot " + experimental_job.fullDisplayName + " please."
  } else {
    println "No experimental match for " + it.fullDisplayName 
  }
}
```

this is only helpful for people on staff who can login to see the script console though

mwoehlke-kitware, 2023-09-25:
Less dirty solution, only needs anonymous access to Jenkins:
```python
#!/usr/bin/env python3

import json
import urllib.request

SERVER = 'https://drake-jenkins.csail.mit.edu/'
TRACKS = [
    'continuous',
    'nightly',
    'weekly',
]

def get_data(*path_components):
    uri = '/'.join([SERVER, *path_components, 'api', 'json'])
    with urllib.request.urlopen(uri) as resource:
        return json.load(resource)

def get_jobs(category):
    jobs = []
    info = get_data('view', category)
    for j in info['jobs']:
        jobs.append(j['name'])

    return set(jobs)

def muge_job_name(name):
    parts = name.split('-')
    for i in range(len(parts)):
        if parts[i] in TRACKS:
            parts[i] = 'experimental'

    return '-'.join(parts)

def main():
    pjobs = get_jobs('Production')
    ejobs = get_jobs('Experimental')

    for pj in pjobs:
        ej = muge_job_name(pj)
        if ej in ejobs:
            print(f'@drake-jenkins-bot {ej} please')

if __name__ == '__main__':
    main()
```

Developers can run this on their machines, or we can have it run according to some logic and push the results somewhere.

jwnimmer-tri, 2023-09-26:
Looks great.

> ... we can have it run according to some logic and push the results somewhere.

Yes, this is the key part of the request.  Its output should be a we page that's hyperlinked from our https://drake.mit.edu/jenkins.html docs.

If nothing else, we could commit the results of the script into https://github.com/RobotLocomotion/drake/tree/master/doc/_pages using pull requests, with some advice/reminders about when to refresh it.  That would add some busywork, but it _would_ give us a nice `git log` / `git blame` historical record of what our set of builds were at any moment in time.

mwoehlke-kitware, 2023-09-26:
For now, these are at:

- https://raw.githubusercontent.com/RobotLocomotion/drake/jenkins-jobs-experimental/request-jobs-provisioned.txt
- https://raw.githubusercontent.com/RobotLocomotion/drake/jenkins-jobs-experimental/request-jobs-unprovisioned.txt

I see two options for keeping the lists up-to-date:

1. Use an Action on drake-jenkins-jobs.
    - **Pros:**
      - Automatic updates only "as needed". (I don't think it's clever enough to know when no changes have occurred, but that's not terrible.)
      - We get a history of our jobs.
    - **Cons:**
      - The Action needs a Secret that allows it to write to the main repository. (Either an SSH deploy key or an API access token.)
2. Generate them while building the documentation.
    - **Pros:**
      - Easy to implement.
    - **Cons:**
      - No history.
      - Queries Jenkins every night.

@jwnimmer-tri, WDYT?

jwnimmer-tri, 2023-09-26:
> We get a history of our jobs.

To be clear, this is a bit of a nice-to-have, not a fundamental requirement.  In theory, the git history of `drake-jenkins-jobs` can get us what we need, in a pinch.

> The Action needs a Secret that allows it to write to the main repository.

This seems surmountable.  As long as the token only has code push permission (versus admin/api permission), it seems safe enough.  We have branch protection turned on for the things we care about.

In the extreme, we could have the 'bot open a pull request (so we can review it as a sanity-check) and that might reduce the permission scope as a bonus.

> Queries Jenkins every night.

I'm not too worried about Jenkins doing this -- it's not really a much of downside.

The actual downside on the front would be if developers who were building the docs locally (to preview how they look) were to start touching the internet.  We should maintain the invariant that by default, local developer preview should _not_ interact with Jenkins.  (It's OK to have that page missing in the docs, in that case.)

---

As far as what to choose?  Whichever one is less work.

BetsyMcPhail, 2023-09-26:
I would also propose, in the interest of simplicity,

3. Whenever drake jenkins jobs is updated, the author also manually runs the script and creates a  drake PR.  The steps left here would be to commit the script and the initial pages and update the drake jenkins jobs internal documentation to reflect the new process.

mwoehlke-kitware, 2023-09-26:
Hehe, I was just adding that to the original comment. 🙂

jwnimmer-tri, 2023-09-26:
Y'all have a better sense of the mechanics that myself.  If an extra manual step or two is satisfactory to you, that works for me too.

mwoehlke-kitware, 2023-09-26:
>> The Action needs a Secret that allows it to write to the main repository.
>
> This seems surmountable.

Right. The point there was more getting permission before just doing it. As far as I can tell, however, the Action either has write permission or it doesn't. "Deploy keys with write access can perform the same actions as an organization member with admin access."

TBH, if we're worried about permissions leaking, an easy fix is to make a new repo that only has the job lists.

PR #20293 (2023-10-02): [doc] Add links to experimental Jenkins bot commands
Resolves #19860

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20293)
closed 2023-10-02 by jwnimmer-tri