RobotLocomotion/drake #19865: Upgrade Externals August 2023

jwnimmer-tri, 2023-07-29:
Follow [Semi-automated monthly upgrades](https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/README.md#semi-automated-monthly-upgrades) documentation.

Prior issue (July 2023): #19715.

BetsyMcPhail, 2023-08-02:
List of updates for this month:

- [ ] abseil_cpp_internal needs upgrade from b8ebbc2ecadb64239dc0767b55296ab9c4510b0f to d2de53124ec55390ce1ec66fd1fbf28b5a7ec4ec
- [ ] common_robotics_utilities needs upgrade from d82b404cf097d52ffa67955f2d57687d1851d4fe to f4c8d5538ed9a9ec26d4c76a059f282d70c69529
- [ ] curl_internal needs upgrade from curl-8_1_2 to curl-8_2_1
- [ ] dm_control_internal needs upgrade from ccbaf9193ea147ad963558549202299ac07c69dc to d6f9cb4e4a616d1e1d3bd8944bc89541434f1d49
- [ ] gtest needs upgrade from v1.13.0 to v1.14.0
- [ ] optitrack_driver needs upgrade from f9dab231878f612db5d7df45f1f9faf08d7a4319 to 8ade24ceb3f8ee10f7e0ae17d7cd55119da72873
- [ ] petsc needs upgrade from v3.19.3 to v3.19.4
- [ ] platforms needs upgrade from 0.0.6 to 0.0.7
- [ ] pycodestyle needs upgrade from 2.10.0 to 2.11.0
- [ ] voxelized_geometry_tools needs upgrade from 6fd994565151764d9099819c3b40033179e5e517 to 37122a2d1c7da410965ab0d800b1dbc1c3ffd5fa

BetsyMcPhail, 2023-08-02:
@jwnimmer-tri before I get too far into this task, I notice that the dependency on some of these packages (e.g. `petsc`) have been deprecated in #17231. Are there packages I should skip? Or carry on with update for everything listed?

jwnimmer-tri, 2023-08-02:
I would say we should add `@petsc` to the `_IGNORED_REPOSITORIES`:
https://github.com/RobotLocomotion/drake/blob/0d9c419a256163a1ae9a03300d0583f1c734cffa/tools/workspace/new_release.py#L54

... because traditionally it has been difficult to upgrade.  (That would be fine as a distinct commit on the multi-commit, all-upgrades PR, if you like.)

The rest I would say let's just upgrade them under the usual process.

closed 2023-08-07 by jwnimmer-tri