RobotLocomotion/drake #19870: Port hydroelastic examples to meldis and/or meshcat

jwnimmer-tri, 2023-07-29:
TBD

jwnimmer-tri, 2023-07-31:
Reminder: Some of the `--help` strings also encode details about `drake_visualizer`.  Those should also be fixed along with the README files.

jwnimmer-tri, 2023-09-12:
In #20175, I've fixed the `ball_plate`.  It seems like Meldis works fine now.  I'll follow up and fix the rest, later on.

DamrongGuoy, 2023-09-20:
The `examples/hydroelastic/spatula_slip_control` uses Meshcat already. Its README instructs users to use `http://localhost:7000/`.  I think now we have two examples left:

- `example/hydroelastic/python_ball_paddle`
- `example/hydroelastic/python_nonconvex_mesh`

I'm experimenting with `example/hydroelastic/python_nonconvex_mesh` in draft PR #20220

jwnimmer-tri, 2023-09-20:
The `spatula_slip_control` still has this in its README:

```
**WARNING**: According to issue
[17681](https://github.com/RobotLocomotion/drake/issues/17681),
only the legacy `//tools:drake_visualizer` application of days past, not
MeshCat nor meldis, can draw wireframe of the contact surfaces.
As a result, you will need `//tools:drake_visualizer` to visualize the 
difference between the `triangle` option and the default `polygon` option.
However, `//tools:drake_visualizer` is only available in the older
Ubuntu Focal, not Jammy.
```

This ticket was intended to also encompass fixing the documentation such as READMEs, gflags help strings, etc.

closed 2024-03-20 by jwnimmer-tri