RobotLocomotion/drake #19901: Expand what is configurable in CameraConfig

SeanCurtis-TRI, 2023-08-01:
This is a meta issue to collect various features that would improve the power of configuring rendering via .yaml files.

- [ ] Label images
    - Currently, a camera can be configured to create color and/or depth images. But never label images.
    - We should add a field `label` that has the same semantics as `depth`: default value is `false`, and setting it `true` creates label images.
- [ ] Image outputs
    - Currently, images are only configured to be broadcast over LCM. That, however, is not the only viable destination for those images.
    - [ ] ImageWriter (write out an image sequence).
    - [ ] VideoWriter (write out the video directly). This would require a C++ conversion of the python-only `VideoWriter` and/or a special patch so that when running in python, the system gets instantiated.

SeanCurtis-TRI, 2023-08-02:
I hadn't looked yet, but I was wondering how the video writer handled its completion.

I don't supposed python gives us any promises that would allow us to automatically call save upon destruction, or some such thing?

SeanCurtis-TRI, 2023-08-02:
Here's a wild and crazy idea.

Based on the [warning on `__del__`](https://docs.python.org/3/reference/datamodel.html#object.__del__), we could have each instance of `VideoWriter` register itself with a `_NamedModuleGlobal`. That global could call `Save()` on each of the registered instances upon deletion. It would be guaranteed to still have PIL/cv2 available.

jwnimmer-tri, 2023-08-02:
Finalizers are somewhat sketchy to rely on, in general.

It's also not the case that "the simulation has finished, please write a video" is the same time as "you've deleted your diagram" or "your interpreter is about to exist".

Perhaps more suitable, we could have a static member function (e.g., `VideoWriter.SaveAll()`) that called `Save()` on all not-yet-saved instances might be suitable.  It would be a one-line in the user's main simulation program, but that's no too bad.

The other line of design thinking would be to somehow get the simulator's events system to post an event to `VideoWriter` that could cause it to save.  I'm not sure if we can do that it any useful way yet, though.

SeanCurtis-TRI, 2023-08-02:
Yeah, the language on `__del__` might be ambiguous. Promising that A gets deleted before B isn't the same as promising that `__del__` gets called on either one.

It's too bad the `VideoWriter` can't do a "valid video up to this point" kind of thing. :(

As for the event system, I'm unsure what that event would be. If you can't call `VideoWriter.Save()` safely multiple times, then it'd have to be an event that promises to fire off once and only once at the end. And knowing the simulator is at the end is *exactly* the problem.

I guess documentation can help.

    - If you're likely to hit Ctrl-C before the end of the solution, but would still like a record of the renderings, use `ImageWriter` and `ffmpeg`.
    - If you're confident the simulation will run its course and you're looking for convenience, use `VideoWriter`.

jwnimmer-tri, 2023-08-02:
Yeah.  If the writer could offer a `video_writer.Checkpoint()` function (in addition to `Save()`), then we could schedule that to run periodically.  Such a function might even be possible, I didn't bother trying to figure it out when I first made `VideoWriter`.  It might also depend on how expensive the checkpointing would get -- whether it was O(N) frames since the last checkpoint (good) or O(N) in the total number of frames (bad).

jwnimmer-tri, 2023-11-06:
I split #20483 into its own issue.