RobotLocomotion/drake #19935: drake_model compatibility with meshcat

SeanCurtis-TRI, 2023-08-03:
### What happened?

Recently, someone tried running atlas and observed that [while the atlas robot used to show textures, it does not do so in meshcat](https://drakedevelopers.slack.com/archives/C43KX47A9/p1691096799659759).  The atlas data is stored as .vtp (polygonal data for VTK) -- e.g., [the torso](https://github.com/RobotLocomotion/models/blob/master/atlas/meshes/mtorso/mtorso_0.vtp) --  and the three.js [VTKLoader.js](https://github.com/mrdoob/three.js/blob/dev/examples/jsm/loaders/VTKLoader.js) appears to read only a portion of the available data, omitting texture data.

Upon a casual perusal, it seems that it's not the only model that won't work meshcat out of the box. For example, the franka data has objs [with more than one material](https://github.com/RobotLocomotion/models/blob/master/franka_description/meshes/visual/finger.obj).

We need to do a survey of which meshes in RobotLocmotion/models that won't play nicely with meshcat and update them.

- [ ] atlas: has .vtp files.
- [ ] franka: has .obj files with too many materials
- [ ] others TBD

jwnimmer-tri, 2023-08-31:
See also #19992.  For Atlas, we should probably purge the robotiq meshes (if any), instead of converting them.

jwnimmer-tri, 2024-03-28:
> The atlas data is stored as .vtp ...

We make no promises about the quality of our model files; they are not part of our [Stable API](https://drake.mit.edu/stable.html) footprint.  Therefore, lacking textures is not a bug that the Drake team plans to fix.  Users who are working with Atlas are invited to submit pull requests to https://github.com/RobotLocomotion/models with improvements, and we can merge them.

>  franka: has .obj files with too many materials

I've re-located this bug report to the models repo: https://github.com/RobotLocomotion/models/issues/47.

> others TBD

No reason to keep a speculative bug TBD open.  If there are bugs that are important, someone will report them acutely.  If not, then it didn't matter anyway.

closed 2024-03-28 by jwnimmer-tri