RobotLocomotion/drake #19959: meshcat backgrounds are broken for orthographic cameras

RussTedrake, 2023-08-08:
### What happened?

```
meshcat = StartMeshcat()
meshcat.Set2dRenderMode(X_WC=RigidTransform(RollPitchYaw(-np.pi / 2, 0, 0),
                                            [0, 0, 1]),
                        xmin=-2,
                        xmax=2,
                        ymin=-2,
                        ymax=2)
meshcat.SetProperty("/Background", "visible", True)
```
produces
[image]

Expected: the entire background should have the background color.

SeanCurtis-TRI, 2023-08-10:
Root cause solved in: https://github.com/rdeits/meshcat/pull/150.

SeanCurtis-TRI, 2023-09-07:
Resolved when meshcat was updated in #20000.

closed 2023-09-07 by SeanCurtis-TRI