RobotLocomotion/drake #19965: Properly test VTK ABI Namespaces

BetsyMcPhail, 2023-08-08:
When VTK ABI namespaces were implemented, (https://gitlab.kitware.com/vtk/vtk/-/merge_requests/8993 / https://github.com/RobotLocomotion/drake/issues/16502), some basic testing was added but it has proven to be insufficient.

There is an open VTK MR to add proper testing to CI (https://gitlab.kitware.com/vtk/vtk/-/merge_requests/9891). _This_ issue is to allow us to add a tracking issue to the drake board for that work.

jwnimmer-tri, 2023-08-30:
The namespace selection does not affect the `vtksys` module.  I think I can working around that locally in Drake with a few patches and settings, but it makes the upstream feature practically useless.  Everything depends on `vtksys`, so if it has ODR violations, nothing else will work.  As part of this task,Iit might be worth investing some time upstream to see if we can get `vtksys` to play our game, too.

jwnimmer-tri, 2023-08-30:
> Do you want me to look at the PR too?

Yes, but for efficiency possibly wait until Sean and I have done a first cycle of comments and fixes, so that we don't hit the same things twice.  So maybe in a day or two.  I probably still have a day or so of manual packaging testing, also.

> I should have enough info to discuss in the meeting tomorrow.

I'm always happy to sync in the f2f meeting, but for discussing technical decisions it would be better to post to the issue(s) first with the information, so we I can digest it ahead of time.  We've been slipping a bit lately into technical deep dives in the f2f sync, without laying the groundwork first by accumulating evidence in written form.  It's always a balance (waiting longer by trying to invest in too much writing has its own risks).  A rule of thumb might be that if it takes more than 2 minutes to explain in a meeting, it should instead be summarized in writing ahead of time.

In any case, upstreaming the vtksys feature is not urgent (it won't block my PRs).  That's the benefit of building our externals from source -- I can hack and slash, without blocking on usptreams.

svenevs, 2023-08-30:
Absolutely, I should be able to summarize what we think it'll look like here on this thread when I get a little more information. I expect to be able to have that summary before tomorrow afternoon. Waiting on some responses currently.

jwnimmer-tri, 2023-10-12:
Per f2f today:

(1) We'll wait for @svenevs to be available before we do much more work.  @mwoehlke-kitware might still tidy up a few loose ends.

(2) The most important near-term goal is to have the C++ code operating correctly.  Drake only uses VTK C++, not Python bindings or anything else.  The original merge request from weeks ago looks like it's trying to enable the namespace testing in a multi-language build (Python, etc).  To the extent that there are challenges with getting Python to work, we should limit the first milestone to have _only_ the C++ code be _compiled_ with the new namespace, and merge that test into master so that no more C++ code goes in without proper namespacing.  After that's locked in place, we can revisit the question of Python + etc support.

(3) My [comment above](https://github.com/RobotLocomotion/drake/issues/19965#issuecomment-1699436987) about vtksys was off-topic for this issue, sorry.  The only goal here is to lock in the namespacing for the normal VTK modules, not for anything third-party nor vtksys.  Likewise, the goal is _not_ that all symbols are perfectly mangled.  If there are C ABI symbols that aren't fixed yet, that's fine.

svenevs, 2023-11-03:
https://gitlab.kitware.com/vtk/vtk/-/merge_requests/10676 was opened, CI will take a little while in the first run, but getting something into CI that will crash developers when they haven't been testing with the ABI namespace is what that solves.

After we get that through (likely: next week), we will file a VTK issue about what the remaining issues are so that we can see what is or is not relevant to this project.

jwnimmer-tri, 2023-11-03:
Looking at the MR, is the only reason that some modules are being disabled at configure-time because the `CheckSymbolMangling.py` test fails?  If so, then it would be _so much better_ if you could instead mark that test as "skip" or "expect-fail" for those modules, instead of disabling the whole module.  Drake primarily cares that everything _compiles_ when the custom namespace is enabled.  For whether mangling was _effective_ Drake has our own tests, and easy ways to work around it.  Drake doesn't necessarily have easy ways to fix build errors.  The most urgent goal is to have VTK CI flag build errors, with all modules enabled.  Fixing the "check symbol mangling" test failures one by one can be done as follow-up.

jwnimmer-tri, 2023-11-03:
So if the goal is actually just to catch build errors, why does the MR disable certain modules?  Do they not even _build_ correctly yet?

```
# These libraries are not supported right now.
set(VTK_MODULE_USE_EXTERNAL_VTK_libharu OFF CACHE BOOL "")
set(VTK_MODULE_USE_EXTERNAL_VTK_exprtk OFF CACHE BOOL "")
set(VTK_MODULE_USE_EXTERNAL_VTK_ioss OFF CACHE BOOL "")
set(VTK_MODULE_USE_EXTERNAL_VTK_fmt OFF CACHE BOOL "")
set(VTK_MODULE_USE_EXTERNAL_VTK_fast_float OFF CACHE BOOL "")
set(VTK_MODULE_USE_EXTERNAL_VTK_verdict OFF CACHE BOOL "")
set(VTK_MODULE_USE_EXTERNAL_VTK_token OFF CACHE BOOL "")
```

> With that macro, we can adequately mangle third party C libraries such as glew (I believe, we haven't looked but seems reasonable). Drake can stop patching. We can isolate working on the ones drake cares about too.

VTK's vendored copies of third-party modules are generally old (seriously backdated) and buggy, and my plan is for Drake to stop using them.  (I've already fixed this for jpeg, png, etc.)  Ultimately Drake is not going to use any of them, rather building all third-party code on our own, with our own vendoring, and our own version pinning.  So, it is of no benefit to Drake for VTK to try to mangle the third-party deps, especially the C ABI third-party deps.

kwryankrattiger, 2023-11-03:
Those aren't disabling modules, it is saying to use the VTK vendored version of certain third-party packages that are not available in the CI container.

jwnimmer-tri, 2023-11-03:
Aha!  Now that I read the list more carefully, I see that's right.  The comment misled me.  I assume it was copy-pasted from somewhere else with more context, where it made sense.

So the simple answer to my:

> Looking at the MR, is the only reason that some modules are being disabled at configure-time because the CheckSymbolMangling.py test fails?

... would be: "nothing is being disabled", so the rest of the thread is moot.

In other words, it seems like the MR is already doing what I asked about.

svenevs, 2023-12-22:
https://gitlab.kitware.com/vtk/vtk/-/merge_requests/10676 is looking much better now, we identified why CI was passing (when it should have been failing), Ryan is going to be finishing the rest of that work off soon.  Thanks Ryan!

svenevs, 2024-01-10:
https://gitlab.kitware.com/vtk/vtk/-/merge_requests/10676 has been merged and the symbols that matter to drake now can be adopted whenever we like, or after the next release.

Docs for flagging the next VTK upgrade in #20763.

jwnimmer-tri, 2024-01-18:
We'll call this finished now that VTK CI has nice tests running all the time.

PR #20765 (2024-01-30): [workspace] Upgrade VTK to latest commit
Closes #19965, #20763.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20765)