RobotLocomotion/drake #19983: Cannot visualise more than one frame added to a body via `AddFrame`

akhilsathuluri, 2023-08-11:
### What happened?

I added a bunch of frames (bodyless, geometryless) attached to a single body via, `AddFrame(FixedOffsetFrame(...))`. Once the frames are added I tried visualising them using `AddFrameTriadIllustration`, which I couldnt since it needs a `FrameId` associated with a `GeometryId`. 

So I fell back to the `AddMultibodyTriad` function from the [manipulation](https://github.com/RussTedrake/manipulation) repo. 

However, this function can only visualise one frame per body since the "first visualised frame" registers a non-unique name with the scene graph and visualising the next frame throws the following error, 

```
RuntimeError: The name 'frame x-axis' has already been used by a geometry with the 'illustration' role.
```

### Quick fix
A simple fix is to edit the `AddTriad` function inside the `AddMultibodyTriad` function to register unique axes names on each call, i.e., instead of 
```
    geom = GeometryInstance(
        X_FT.multiply(X_TG), Cylinder(radius, length), name + " x-axis",
    )
```
it could do,
```
    geom = GeometryInstance(
        X_FT.multiply(X_TG), Cylinder(radius, length), name + " x-axis"+str(unique_id),
    )
```
and then one can visualise as many triads per body as needed. 

### Version

drake 1.15.0

### What operating system are you using?

Ubuntu 22.04

jwnimmer-tri, 2023-08-11:
> I fell back to the AddMultibodyTriad function from the [manipulation](https://github.com/RussTedrake/manipulation) repo.  However, this function can only visualise one frame per body ...

That would be a missing feature wrt the `manipulation` repo, not a problem with Drake.

For Drake, I think it would be better to solve the original problem:

> I tried visualising them using AddFrameTriadIllustration, which I couldnt since it needs a FrameId associated with a GeometryId.

For this problem, would a solution be to enhance [pydrake.visualization.AddFrameTriadIllustration](https://drake.mit.edu/pydrake/pydrake.visualization.html#pydrake.visualization.AddFrameTriadIllustration) to accept a [pydrake.multibody.tree.Frame](https://drake.mit.edu/pydrake/pydrake.multibody.tree.html#pydrake.multibody.tree.Frame) reference (in addition to the `body` or `frame_id` currently supported)?

In other words, this change to the function signature:
```patch
--- a/bindings/pydrake/visualization/_triad.py
+++ b/bindings/pydrake/visualization/_triad.py
@@ -10,13 +10,14 @@ from pydrake.geometry import (
 )
 from pydrake.math import RigidTransform
 from pydrake.multibody.plant import MultibodyPlant
-from pydrake.multibody.tree import Body
+from pydrake.multibody.tree import Body, Frame
 
 
 def AddFrameTriadIllustration(
     *,
     scene_graph: SceneGraph,
     body: Body = None,
+    frame: Frame = None,
     frame_id: FrameId = None,
     plant: MultibodyPlant = None,
     name: str = "frame",
@@ -33,10 +34,12 @@ def AddFrameTriadIllustration(
     Args:
       scene_graph: the SceneGraph where geometry will be added.
       body: when provided, illustrates the frame of the given body;
-        either body= or frame_id= must be provided, but not both.
+        exactly one of body= or frame= or frame_id= must be provided.
+      frame: when provided, illustrates the given multibody frame;
+        exactly one of body= or frame= or frame_id= must be provided.
       frame_id: when provided, illustrates the given geometry.FrameId
         registered with the given plant and scene_graph;
-        either body= or frame_id= must be provided, but not both.
+        exactly one of body= or frame= or frame_id= must be provided.
       plant: MultibodyPlant associated with the given frame_id;
         required if and only if a frame_id= is being used (not body=).
       name: the added geometries will have names "{name} x-axis", etc.
```

akhilsathuluri, 2023-08-11:
> That would be a missing feature wrt the manipulation repo, not a problem with Drake.

Definitely. 

Yes, addiing an additional argument to handle a `frame` should do it.
```
 def AddFrameTriadIllustration(
+    frame: Frame = None, ...
```

Thanks for the quick reply. I should be able to fix this.

akhilsathuluri, 2023-08-12:
Im just wondering if doing as suggested is even possible, since `RegisterGeometry` in the `AddFrameTriadIllustration` requires a `frame_id`. I tried to get one by registering the frame created by `AddFrame` using `RegisterFrame`, which wasnt possible as it expects a `GeometryFrame`. Seems like there is no straigtforward way to visualise a non-geometry frame currently, or am I missing something?

jwnimmer-tri, 2023-08-12:
Yes, without #10247 it is a bit more nuanced.

I think the approach would be similar to this:
https://github.com/RobotLocomotion/drake/blob/57e044df411635a63703cb5e88314dc7be307418/systems/sensors/sim_rgbd_sensor.cc#L21C1-L41

If the multibody Frame is a BodyFrame, then the Body will have a geometry FrameId available, so we can use that.

Or instead, if the multibody Frame is a FixedOffsetFrame, then we need find its parent BodyFrame and offset the triad geometry by the same amount.

rpoyner-tri, 2023-08-15:
Assigning @joemasterjohn per https://drake.mit.edu/platform_reviewer_checklist.html

jwnimmer-tri, 2024-02-15:
This has become more relevant to us recently.

Let's say this as the victory condition: that [model_visualizer](https://drake.mit.edu/pydrake/pydrake.visualization.model_visualizer.html) shows fixed offset frames (e.g., parsed from SDFormat `<frame>` tag) when `--visualize_frames` is enabled.

Under the hood, that will mean that `AddFrameTriadIllustration` gains a third kind of thing it can show -- a `frame: pydrake.multibody.tree.Frame`.  Still only (and exactly) one of `frame_id` or `body` or `frame` can be passed as a kwarg.

If the Frame is a Body frame, we can just show the body.  If it's a FixedOffsetFrame, then we need to back out the offset transform.

Random idea: when `model_visualizer` is calling `AddFrameTriadIllustration` in a loop, right now it uses a 3.0 multiplier for the world frame (to make it bigger). You could imagine that showing fixed offset frames with a 0.75 or something, so that it's a little easier to grok body frames from offset frames at a glance.

jwnimmer-tri, 2024-02-29:
With #21064, the triad logic is able to support this use case case now.  The remaining work in this ticket is to update `model_visualizer` to loop over the offset frames and add them as well.

PR #21318 (2024-04-24): ModelVisualizer draw all MultibodyPlant frames when requested
Rather than visualizing the frames registered with scene graph, it visualizes all frames with a fixed offset w.r.t. the corresponding body.

In support of that, some further changes have been made:

- MeshcatVisualizer now makes use of geometry *name* and geometry id to produce paths.
- AddFrameTriadIllustration() takes a new "brightness" parameters to use brightness as a means to distinguish different kinds of frames.

Closes #19983.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/21318)
closed 2024-04-24 by jwnimmer-tri