RobotLocomotion/drake #19992: Atlas models that don't parse

jwnimmer-tri, 2023-08-14:
As of current Drake (v1.19.0), the `package//drake_models/atlas/...` files have various problems:

These ones don't even parse at all:
- package://drake_models/atlas/robotiq.urdf
- package://drake_models/atlas/robotiq_tendons.urdf

These ones have warnings:
- package://drake_models/atlas/atlas_convex_hull.urdf
- package://drake_models/atlas/atlas_minimal_contact.urdf
- package://drake_models/atlas/robotiq_simple.urdf

I'm not sure that anything here is worth fixing, but maybe it's better to just delete the broken models instead of ship them.

Maybe delete all of the robotiq models, and then fix the warnings on atlas?

\CC @rpoyner-tri @amcastro-tri for any suggestions.

rpoyner-tri, 2024-01-31:
https://github.com/RobotLocomotion/models/pull/30 chips away some problems.

jwnimmer-tri, 2024-03-20:
I no longer think we need a tracking issue for this.  Once we have a model linter per #17061, then we'll need to fix its complaints.  If we don't have a linter, then there's no need to fix anything.

closed 2024-03-20 by jwnimmer-tri