RobotLocomotion/drake #19993: Trajectory base class function for taking blocks

jwnimmer-tri, 2023-08-15:
**Is your feature request related to a problem? Please describe.**

Many times, motion plans will use the `Trajectory` class to specify the path for a multi-dof robot.  Sometimes, we'll want to "slice" that trajectory by dof, and transmit certain dofs to individual command drivers (e.g., hand, vs arm, vs torso).

Some concrete Trajectory classes already support this "matrix block" operation on their values (PiecewisePolynomial and BsplineTrajectory), but there is no _generic_ way to do this.

The request here is for a generic way (in the base class) to accomplish this.

**Describe the solution you'd like**

A new public function on the base class, something like this.

```c++
class Trajectory {
  // ...

  /**
   * Extracts a trajectory representing a block of size (block_rows, block_cols)
   * starting at (start_row, start_col) from this Trajectory.
   *
   * @returns a new Trajectory such that
   *   ret.value(t) = this.value(t).block(i,j,p,q);
   * are only a portion of this trajectory's values.
   * The runtime type of the result will be the same as the type of this.
   *
   * @throws std::exception if this subclass of Trajectory does not support
   * efficient slicing.
   */
  std::unique_ptr<Trajectory<T>> MakeBlock(
      int start_row, int start_col, int block_rows, int block_cols) const;

  // ...
```

**Describe alternatives you've considered**

We could offer a generic `TrajectoryView` decorator whose `value()` function called `value()` on the wrapped trajectory and then copied the block part it wanted. 

This would be maximally compatible, but would waste computation in many casesl.

RussTedrake, 2023-08-16:
fyi -- we do have this capability for PiecewisePolynomial:
https://drake.mit.edu/doxygen_cxx/classdrake_1_1trajectories_1_1_piecewise_polynomial.html#a45001b70f1a92e1d01778b69bb61adf3

jwnimmer-tri, 2023-08-16:
Right.  The goal here is to add a new base class method that calls that derived class method (in subclasses where it exists), so that users don't need to downcast to call it.