RobotLocomotion/drake #19996: Return variable mapping in MakeSemidefiniteRelaxation

RussTedrake, 2023-08-15:
Request: Callers of MakeSemidefiniteRelaxation(prog) should get a return value that let's them relatively easily find the new decision variables that correspond to combinations of the original decision variables.

The Draft PR https://github.com/RobotLocomotion/drake/pull/19886 proposed one idea, and collected some good feedback.  We like the idea of returning a map (or function) that goes from monomials => decision variables.

cc @hongkai-dai , @bernhardpg , @lujieyang

RussTedrake, 2024-12-13:
@AlexandreAmice -- wdyt? is this still needed/requested?  or can i close it?

AlexandreAmice, 2024-12-14:
I think `MakeSemidefiniteRelaxation` is somewhat tailored to the first order relaxation. In this case, getting the entry in X corresponding to some product `x_{i}*x_{j}` should be as easy as finding the index of `x_{i}` and `x_{j}` in the last column and then using those indices in `X`. 

I suppose we could make this easier by returning the map from the original program variables to the row index in `X` which would be almost no work because this map (`variables_to_sorted_indices`) is already created in `DoMakeSemidefiniteRelaxation`.

That being said, I am not sure I understand the use case for this method.

RussTedrake, 2024-12-15:
@lujieyang and @bernhardpg were requesting it right after i wrote the first version.  can you guys fill in the use case for us?

bernhardpg, 2024-12-17:
It would be slightly more complicated in the case where one uses variable grouping. But I agree, it is not super hard to write yourself.

I think it would be a nice-to-have utility functionality, but I also think that most people who would want this would be capable of picking out the PSD matrices from the generated PSD constraints themselves and then finding the right monomial terms (as long as the PSD constraints follow the variable group ordering and there is no other unexpected reordering happing in that case? @AlexandreAmice )