RobotLocomotion/drake #19998: //bindings/pydrake/visualization:py/model_visualizer_test test_reload failure in CI

BetsyMcPhail, 2023-08-15:
### What happened?

 **//bindings/pydrake/visualization:py/model_visualizer_test**  failed in CI:

Aug 7 (x2)
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-monterey-clang-bazel-continuous-release/1101/
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-monterey-clang-bazel-continuous-release/1102/

Aug 15 
https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-arm-ventura-clang-bazel-continuous-release/345/

### What operating system are you using?

Other

### Relevant log output

```shell
======================================================================
FAIL [1.205s]: test_reload (model_visualizer_test.TestModelVisualizer.test_reload)
Checks that the _reload() function does not crash.
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/Users/admin/workspace/mac-arm-monterey-clang-bazel-continuous-release/_bazel_admin/d8209b58b645930155068a99d03380e4/sandbox/darwin-sandbox/7614/execroot/drake/bazel-out/darwin_arm64-opt/bin/bindings/pydrake/visualization/py/model_visualizer_test.runfiles/drake/bindings/pydrake/visualization/test/model_visualizer_test.py", line 229, in test_reload
    self.assertEqual(meshcat.GetButtonClicks(button), 1)
AssertionError: 0 != 1

----------------------------------------------------------------------
```

BetsyMcPhail, 2023-08-15:
Closing this issue for now, it can be re-opened if needed.

BetsyMcPhail, 2023-08-16:
8/15 https://drake-jenkins.csail.mit.edu/job/mac-arm-monterey-clang-bazel-continuous-release/1134/

BetsyMcPhail, 2023-08-16:
8/16 https://drake-jenkins.csail.mit.edu/job/mac-arm-ventura-clang-bazel-continuous-release/353/

BetsyMcPhail, 2023-08-21:
8/19: https://drake-jenkins.csail.mit.edu/job/mac-arm-monterey-unprovisioned-clang-bazel-nightly-release/323/

ggould-tri, 2023-08-30:
Again:
https://drake-jenkins.csail.mit.edu/view/Production/job/mac-arm-monterey-unprovisioned-clang-bazel-nightly-release/334/
Reopening -- it's definitely recurred.

sherm1, 2023-08-30:
Assigning to @BetsyMcPhail for disposition.

svenevs, 2023-09-05:
2023-09-05: https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-unprovisioned-clang-bazel-nightly-release/114/consoleFull

BetsyMcPhail, 2023-09-14:
Reopening as this appeared again on 9/14 --> https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-clang-bazel-nightly-everything-release/118/

jwnimmer-tri, 2023-10-09:
I'm not sure that "priority: high" was appropriate for a one-time flake, but anyway => #20337 for the next attempt at a fix.

svenevs, 2024-01-09:
https://drake-jenkins.csail.mit.edu/view/Nightly%20Production/job/mac-arm-ventura-clang-bazel-nightly-everything-release/241/consoleFull produced this again.  I suppose the plan of action will be reopen and monitor

jwnimmer-tri, 2024-01-09:
Nope.  That log says "FLAKY" not "FAIL".  There was no problem:
```
[1:14:20 AM]  //bindings/pydrake/visualization:py/model_visualizer_reload_test          FLAKY, failed in 1 out of 2 in 10.6s
```

The failures in that build were linter failures.

PR #20129 (2023-09-06): [py visualization] De-flake TestModelVisualizer.test_reload
Closes #19998.

I cannot reproduce the problem locally.  Hopefully it's the kind of thing that will be solved by a bit more sleeping.

---
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20129)
PR #20337 (2023-10-10): [py visualization] Split out flaky visualizer reload test
Closes #19998.

---
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20337)
closed 2024-01-09 by jwnimmer-tri