RobotLocomotion/drake #20006: Geometry benchmark instructions no longer correct

SeanCurtis-TRI, 2023-08-17:
### What happened?

In [`render_benchmark_doxygen.h`](https://github.com/RobotLocomotion/drake/blob/master/geometry/benchmarking/render_benchmark_doxygen.h#L120-L121) we have some specific instructions for running the benchmark.

However, when following those instructions, one receives the following error:

```sh
$ bazel run //geometry/benchmarking:render_experiment -- --output_dir=trial2 --save_image_path="/tmp" --show_window=true
INFO: Running command line: bazel-bin/geometry/benchmarking/render_experiment '--output_dir=trial2' '--save_image_path=/tmp' '--show_window=true'
usage: benchmark_tool.py [-h] --binary BINARY --output_dir OUTPUT-DIR [--sleep SLEEP] [--cputask N]
                         [extra_args ...]
benchmark_tool.py: error: unrecognized arguments: --save_image_path=/tmp --show_window=true
```

rpoyner-tri, 2023-08-21:
So, the incantation that "works" is something like:

```
bazel run //geometry/benchmarking:render_experiment -- --output_dir=trial2 -- -save_image_path="/tmp" -show_window=true
```

However, I have my doubts about how useful that command is, since it runs a bunch of iterations under the benchmark framework, so you end up seeing some long train of randomly appearing and disappearing windows with green spheres in them.

I'll need to read the whole file to figure out what is really wanted.

rpoyner-tri, 2023-08-23:
Closed by #20039.

closed 2023-08-23 by rpoyner-tri