RobotLocomotion/drake #20011: linting doesn't handle "// namespace foo DRAKE_NO_EXPORT" consistently

SeanCurtis-TRI, 2023-08-17:
### What happened?

If we've got some C++ code:

```
namespace foo DRAKE_NO_EXPORT {
...
}  // namespace foo
```

This makes the linter happy. However, the clang-format idempotency test fails because it wants to change it to:

```
namespace foo DRAKE_NO_EXPORT {
...
}  // namespace DRAKE_NO_EXPORT
```

(And my VS code local application makes it `// namespace foo DRAKE_NO_REPORT`, which tells me our clang-format is more than just the .clangformat file? Or I haven't got my VS code configured properly).

But certainly, the two linters should *not* be fighting each other in this regard.

SeanCurtis-TRI, 2023-08-18:
Relates to #4843.

jwnimmer-tri, 2023-08-18:
Putting `FixNamespaceComments: false` in `drake/.clang-format` seems to resolve the problem.  It was nice that it would auto-fix these things for us, but if the autofixing broken then I guess we need to turn it off.  The `cpplint` checks will remind us when something goes wrong, to fix it by hand.

jwnimmer-tri, 2023-08-19:
It's already been filed upstream as https://github.com/llvm/llvm-project/issues/61251.

jwnimmer-tri, 2023-08-19:
After #20028 lands, we could plausibly build `clang-format` from source, so that we can use the fixed version from upstream.  It [already has bazel rules](https://github.com/llvm/llvm-project/blob/llvmorg-16.0.6/utils/bazel/llvm-project-overlay/clang/BUILD.bazel#L2108-L2109), the only caveat would new workspace dependencies that `llvm` also induces.

jwnimmer-tri, 2023-08-19:
I have a prototype of build-from-source here: https://github.com/jwnimmer-tri/drake/commits/clang-format-as-external.  There aren't any new dependencies.  It's only a question of taming the build warnings and build time.

SeanCurtis-TRI, 2023-08-20:
For now, we'll close this with the guidance that we control the formatter around the few affected sites.

```c++
namespace foo DRAKE_NO_EXPORT {
...
// clang-format off
} // namespace foo
// clang-format on
```

Disabling across the board is simply too painful.

closed 2023-08-20 by SeanCurtis-TRI