RobotLocomotion/drake #20014: Consistent declaration and documentation of dependencies in MBP

amcastro-tri, 2023-08-17:
Per @jwnimmer-tri's suggestion in #19920:

> I suggest that as a follow up, for each callback function that we register with the framework (i.e., outputs or cache entries), we consistently document atop the function which tickets its declared to use. Then, we have a three-prong checklist to review, that gives us the full proof we need:
>
> - For each this->Declare(&CalcFunction, tickets) line, the tickets is congruent with the comment atop CalcFunction implementation.
> - For each CalcFunction body, any data it accesses from the context is listed in the comment atop it's implementation.
>  - For each CalcFunction body, any helper functions it calls has the same or subset of tickets -- compare ticket docs atop the caller and callee functions.
>
> The checklist only applies to functions that accept the full Context. Functions that accept narrower data (that consistent with the tickets from each call site) are OK on their own without digging in to them.
```

joemasterjohn, 2023-08-22:
Per #20022, cache entries for hydro computations do not have proper dependencies, namely on the geometry query input port (used for material parameters: friction, dissipation, hydroelastic modulus, etc.). In my (non-exhaustive) effort to get the right dependencies, I only had to add:
```
this->get_geometry_query_input_port().ticket()
```
to the list of dependencies for the `contact_info_and_body_spatial_forces` cache entry.

jwnimmer-tri, 2024-11-02:
In #21623 and its lead-up, we did some what to clean this up.  I'm calling that good enough for now.

closed 2024-11-02 by jwnimmer-tri