RobotLocomotion/drake #20029: common::MaybePauseForUser() hangs for some tests when run under `bazel run`.

RussTedrake, 2023-08-20:
### What happened?

```
bazel test //geometry/optimization:hpolyhedron_test 
```
works as expected.  No pauses.

```
bazel build //geometry/optimization:hpolyhedron_test && bazel-bin/geometry/optimization/hpolyhedron_test
```
works as expected. It pauses for users to see the visualizations.

Unfortunately, 
```
bazel run //geometry/optimization:hpolyhedron_test 
```
hangs. 

I'm not sure exactly the cause.
```
bazel run //geometry:meshcat_manual_test
```
works fine.

### Version

master

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

compiled from source code using Bazel

jwnimmer-tri, 2023-08-20:
The `meshcat_manual_test` is a `cc_binary`.  The `hpolyhedron_test` is a `cc_test`.  That probably accounts for why the two differ.

rpoyner-tri, 2023-08-23:
Closed by #20031.

closed 2023-08-23 by rpoyner-tri