RobotLocomotion/drake #20032: DoGetWitnessFunctions in Python throws RuntimeError: Unable to cast Python instance to C++ type

drewhamiltonasdf, 2023-08-21:
### What happened?

Hi,

I've been trying to put together some sample Hybrid systems using the WitnessFunction class and I am getting what looks like errors related to Pybind. I'm pasting some very minimal code for reference:

```python
class HybridSystem(LeafSystem):
    def __init__(self):
        super().__init__()  # Initialize the base class.

        # Continuous State
        state_index = self.DeclareContinuousState(2)  
        self.DeclareStateOutputPort("y", state_index) 
    
        self.witness = self.MakeWitnessFunction(
            "witness", WitnessFunctionDirection.kCrossesZero,
            self.WITNESSFUNCTION)
        
    def WITNESSFUNCTION(self, context):
        self.called_witness = True
        print("_witness")
    
    def DoCalcTimeDerivatives(self, context, derivatives):
        x = context.get_continuous_state_vector().GetAtIndex(0)
        xd = context.get_continuous_state_vector().GetAtIndex(1)
        derivatives.get_mutable_vector().SetAtIndex(0, xd)
        derivatives.get_mutable_vector().SetAtIndex(1, 0)

    def DoGetWitnessFunctions(self, context):
        self.called_getwitness = True
        return [self.witness] #, self.reset_witness, self.system_reset_witness]
```

This code prints "_witness" once, and then fails on simulator.AdvanceTo(4.0) with "RuntimeError: Unable to cast Python instance to C++ type":

```python
# Construct an instance of this system and a context.
system = HybridSystem()
context = system.CreateDefaultContext()

builder = DiagramBuilder()
builder.AddSystem(system)
logger = LogVectorOutput(system.GetOutputPort('y'), builder)

builder.ExportOutput(system.GetOutputPort('y'))
diagram = builder.Build()

simulator = Simulator(diagram)
sim_context = simulator.get_mutable_context()

# Set the initial conditions:
sim_context.SetTime(0)
sim_context.SetContinuousState([0.5,0])

# Run the simulation forward.
simulator.AdvanceTo(4.0)
```

I was attempting to put together some minimal examples from the book "Impulsive and Hybrid Dynamical Systems" as a set of tutorials I was going to share. As there weren't any examples I could find of using the WitnessFunction python bindings, I ended up looking through the tests folder in Drake to see how to call these. 

Typically, I would struggle my way through, but this smells a bit like a bug, so I thought I'd post it here. (This code is not meant to do anything interesting other than get a working example together).

Best,
Drew

### Version

pydrake-0.2.1

### What operating system are you using?

Ubuntu 22.04

### What installation option are you using?

Unknown / Other

### Relevant log output

```shell
Ubuntu 22.04.3 LTS
```

jwnimmer-tri, 2023-08-21:
The `def WITNESSFUNCTION(self, context)` needs to return a `float` value.  Returning `None` might be the cause of the (less than helpful) exception message.

drewhamiltonasdf, 2023-08-22:
Closing this issue.

@jwnimmer-tri thanks for your reply. I'm blushing. I definitely could have figured that out myself. Though, I do want to point out that, in my case I had to return some non-sense as I was handling a slightly complicated system with return floats inside if-statements.

I also noticed that you have to set the accuracy of the simulation or you get another run-time error: `Integrator is not operating in fixed step mode and accuracy is not set in the context.`

All that said, having a lot of fun digging into the full Python API, and looking forward to being able contribute (just to Python documentation for now).

I put what I came up with (a little unpolished) in https://github.com/drewhamiltonasdf/DrakeTutorials/blob/main/004%20-%20Hybrid%20Plant.ipynb

I'm going to keep these tutorials there and polish them up a bit, with the hope that I can submit a pull request. 

Thanks again for the quick reply. I really love this project.

PR #20106 (2023-09-01): [py systems] Improve error message for missing witness functions
When the user accidentally forgets the `return` statement, the error message was very poor.

Closes #20032.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20106)
closed 2023-08-22 by drewhamiltonasdf