RobotLocomotion/drake #20036: Poor performance of HPolyhedron::UniformSample

markisus, 2023-08-21:
### What happened?

The hit-and-run algorithm only converges to the uniform distribution after a warmup period, aka mixing time. The paper "HIT-AND-RUN FROM A CORNER by Lovasz and Vempala" prove a convergence result (Corollary 1.2) which shows the dependence on the number of steps taken, and closeness to the uniform distribution. 

[image]

As it stands, HPolyhedron::UniformSample uses only a single iteration over a previous sample to return the next sample. If my understanding is correct, there needs to be an inner loop that runs many times, in order for the result to be close to uniformly distributed.

This is apparent if you imagine sampling from a very long, very thin, rod, in very high ambient dimension. For many samples, hit and run will not progress, since it must be very lucky to sample the into the long dimension of the rod. For this case, it's clear that consecutive samples are extremely correlated.

For standalone usage, this is not a big problem, since users can just manually warm up the function. But UniformSample is also used internally in IrisInConfigurationSpace during the counter-example search. For long and thin configuration spaces, I believe this will lead to extremely correlated starting seeds.

markisus, 2023-08-22:
Yes I am able to construct arbitrarily bad counterexamples by making long-thin rectangles in 2d. The skinnier the rectangle, the poorer the performance. Here is a script that generates one million samples from a (10,000, 0.0001) rectangle.

```
from pydrake.all import (
    HPolyhedron,
    RandomGenerator
)
import numpy as np
from matplotlib import pyplot as plt

box_length = 1e4
box_height = 1e-4
long_thin_box = HPolyhedron.MakeBox(
    np.array([0.0, 0.0]),
    np.array([box_length, box_height]))

generator = RandomGenerator()

num_samples = 1000000
sample = long_thin_box.UniformSample(generator)
samples = []
for _ in range(num_samples):
    sample = long_thin_box.UniformSample(generator, sample)
    samples.append(sample)

samples = np.array(samples)
xs = samples[:,0]
ys = samples[:,1]

plt.title("HPolyhedron UniformSample on a Long Thin Rectangle")
plt.plot(xs)
plt.xlabel("Iteration Number")
plt.ylabel("Sample X Coordinate")
plt.show()
```

You can see the poor performance by looking at the x-coordinates of the samples. 
[image]

A performant implementation should see the x coordinates ranging from 0 to 10,000. The Drake implementation just barely makes it up to 80. 

You can also see that there is high correlation between subsequent samples, no matter how long the sampling has been run. You said "It would only have to do this when it is first started, and when it's restarted due to the initialization becoming infeasible." Unfortunately, I think this adversarial example illustrates why that wouldn't work. The mixing must occur in between every call to UniformSample. 

For IrisInconfigurationSpace, I believe that this would be okay, since making the initial guess takes trivial time in comparison with running the subsequent optimization problem. 

In terms of real-world example, I encountered a case like this where I had a robot that had to slide itself into a thin corridor.

cohnt, 2023-08-22:
> For IrisInconfigurationSpace, I believe that this would be okay, since making the initial guess takes trivial time in comparison with running the subsequent optimization problem.

The solves required for the strategy to get a uniform distribution probably isn't trivial. 1e6 samples takes ~.8 seconds on my machine, and isn't doing a lot of mixing. Even 1e8 samples only saw a quarter of the space (see below), and that takes ~80 seconds, which is clearly too much per iteration. So I don't think just cranking up the amount of mixing is going to solve the problem you're facing.

[image]

Note that your runtime is also going to grow a lot if you're facing larger problems. In a 14 dimensional space with 448 hyperplanes, drawing 1e7 samples took ~36 seconds, and achieved very little mixing. Getting nice, uniform samples from an HPolyhedron is hard, especially in these adversarial cases. Rejection sampling isn't going to work either, unless you have a narrow outer approximation. (And realistically, you don't have one of those unless it's axis-aligned.) Initializing samples at the vertices might help if you have them, but converting from HPolyhedron to VPolytope requires exponential time.

However, IRIS-NP doesn't truly need uniform samples to succeed, because these are just the initial guesses for a nonlinear program. As long as you're getting samples in the basin of a feasible local optimum, the solver should be able to find its way. In the real-world example you're looking at, what is the failure case that you're observing? Is IRIS failing to find hyperplanes that separate you from an obstacle? Are you able to share the real-world example?

PR #20974 (2024-02-21): Add mixing_steps to IrisOptions
Resolves #20036.

+@alexandreamice for feature review, please.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20974)
commit 6b349d5e (2024-02-21): Add mixing_steps to IrisOptions (#20974)

Resolves #20036.
closed 2024-02-21 by ggould-tri