RobotLocomotion/drake #20037: Represent Lower-Dimensional Ellipsoids

cohnt, 2023-08-21:
Right now, Drake has the [`Hyperellipsoid`](https://drake.mit.edu/doxygen_cxx/classdrake_1_1geometry_1_1optimization_1_1_hyperellipsoid.html) class, which can be used to represent ellipsoidal convex sets. These sets are allowed to be unbounded, but they cannot be lower dimensional. For methods such as the minimum volume circumscribed ellipsoid (implemented in #19857), it may be necessary to consider such sets.

The documentation for `Hyperellipsoid` already describes how such a set should be formulated, so all that's needed is to create another subclass of `ConvexSet` to support such features. I'll plan to implement this in the coming days.

- [ ] Create `AffineBall` subclass of `ConvexSet`.
- [ ] Add constructors to transform between `AffineBall` and `Hyperellipsoid` if possible.
- [ ] Replace `Hyperellipsoid::MinimumVolumeCircumscribedEllipsoid` with `AffineBall::MinimumVolumeCircumscribedEllipsoid`.

Tagging @RussTedrake and @wernerpe, since this issue is relevant to them.

sadraddini, 2023-08-27:
Late to the conversation because I see the PRs now.

Having two very, very, similar classes is awkward. Some suggestions:
1- Disallow unbounded hyperellipsoids and replace it with the affine transformation notation. How useful are unbounded hyperellipsoids anyways? We can represent them by Cartesian product of a subspace with a  bounded hyper-ellipsoid, as an example.
2- Generalize affine transformations of any convex set as a new dervied class.

RussTedrake, 2023-08-27:
@sadraddini -- do you feel similarly about HPolyhedron and VPolytope.  For me, the analogy to this pair is strong.

sadraddini, 2023-08-27:
VPolytope and HPolyhedron have more differences that merit two separate classes. They have very different PointInSet checks, constraint encodings, etc.
Most properties of Hyperellipsoids and AffineBalls are the same and they only differ in one handling unbounded sets and the other handling lower dimensional sets. 
I think the [commonsense definition](https://en.wikipedia.org/w/index.php?title=Ellipsoid&action=edit&section=23) leans toward bounded sets (we had the same argument about [hyperrectangles](https://github.com/RobotLocomotion/drake/pull/19982) and decided we only consider bounded ones). We can make a ctror / static method to handle unbounded sets by decomposing them into a Cartesian product of a bounded set (AffineBall in this case) and an affine subspace - I feel that is straightforward.

TobiaMarcucci, 2023-09-11:
I agree that there should be some underlying class that all there variations of quadratics point to. To my eyes, this could be a second-order-cone class. But it's also true that different parameterizations of ellipsoids are good for different things. For example, consider the problems of finding the largest ellipsoid contained in a polytope or the smallest ellipsoid that contains a set of points. For the former the convex parameterization is {A x + d : || x || <= 1}, for the other it is {x : || A x + b || <= 1}. So I would not say that is necessarily bad to have different versions of ellipsoids, as long as they share all the sharable pieces of code?

cohnt, 2023-11-07:
Issue is now solved with the completion of #20399

closed 2023-11-07 by cohnt