RobotLocomotion/drake #20045: In deformable contact the definition of contact normal vector is not consistent

zeshunzong, 2023-08-22:
In SAP, given contact normal $\vec{n}$, we require that that relative velocity expressed in $\vec{n}$ as a scalar is negative when the two contact objects are getting closer to each other (or penetrating deeper). 

If we define the relative velocity as $\vec{v}_B - \vec{v}_A$ (as is done in `ContactPairKinematics`), then the contact normal must be defined as pointing from A to B.

`DeformableDriver::AppendDiscreteContactPairs()` computes the normal as pointing from B to A. `DeformableDriver::AppendContactKinematics()` sneaks a minus sign to get the correct normal for solver. This is also not clearly documented either.

The former will result in inconsistency in evaluating forces, which is not done for deformable yet, but should and will be done in `CompliantContactManager::DoCalcContactResults()`

@xuchenhan-tri

EricCousineau-TRI, 2023-08-24:
Assigning @xuchenhan-tri for disposition

xuchenhan-tri, 2024-03-26:
This has been fixed with #20301 and #20913.

closed 2024-03-26 by xuchenhan-tri