RobotLocomotion/drake #20049: IDTO support: Allow access to the geometry id to body index

anrp-tri, 2023-08-23:
IDTO uses this information to extract/derive information to calculate contact force and contact jacobians. I am not sure if it should be exposed, though, as I see that there is a "contact manager" that is supposed to be implemented to "manage" this.

jwnimmer-tri, 2023-08-24:
See https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_multibody_plant.html at "Body frames and SceneGraph frames":

> Given a GeometryId, SceneGraph cannot report what body it is affixed to. It can only report the SceneGraph alias frame F. But the following idiom can report the body:
> ```c++
> const MultibodyPlant<T>& plant = ...;
> const SceneGraphInspector<T>& inspector =  ...;
> const GeometryId g_id = id_from_some_query;
> const FrameId f_id = inspector.GetFrameId(g_id);
> const Body<T>* body = plant.GetBodyFromFrameId(f_id);
> ```

Then `body->index()` will return the `BodyIndex`.

anrp-tri, 2023-08-30:
Yep, that works. Thanks!

closed 2023-08-30 by anrp-tri