RobotLocomotion/drake #20053: [HardwareSim] Can't easily terminate `//examples/hardware_sim:demo_py`

RussTedrake, 2023-08-24:
I'm not able to terminate the sim... for both
```
bazel run //examples/hardware_sim:demo_py
```
and
```
python3 hardware_sim.py --scenario_file=example_scenarios.yaml   --scenario_name=Demo
```
somehow I'm not able to Ctrl-C, and have to manually kill the process.  Is that just me?

jwnimmer-tri, 2023-08-27:
As a workaround, `Ctrl-\` still works (SIGQUIT).

The reason `Ctrl-C` doesn't work is that the simulation is stuck in C++ land, never checking back in with Python.

Here's one way to fix it:

```patch
--- a/examples/hardware_sim/hardware_sim.py
+++ b/examples/hardware_sim/hardware_sim.py
@@ -174,6 +174,9 @@ def run(*, scenario):
     random = RandomGenerator(scenario.random_seed)
     diagram.SetRandomContext(simulator.get_mutable_context(), random)
 
+    # Force a per-step call back into Python, so that Ctrl-C works.
+    simulator.set_monitor(lambda _: None)
+
     # Simulate.
     simulator.AdvanceTo(scenario.simulation_duration)
 
```

I wonder if that's good enough, or if we should try to _automatically_ have long-running C++ functions ping back into Python as part of the bindings themselves?

RussTedrake, 2023-08-27:
Do you think pinging back to python on every step is more than we need?  I guess we have other examples of calling `simulator.AdvanceTo()` in a loop until we reach the simulation duration?

jwnimmer-tri, 2023-09-01:
I tend to dis-prefer patterns that look like this ...

```py
while simulator.get_context().get_time() < scenario.simulation_duration:
    simulator.AdvanceTo(min(scenario.simulation_duration, simulator.get_context().get_time() + 1.0))
```

... because they introduce integrator halts that differ from the underlying dynamics of the diagram.  Sometimes they will even cause really tiny timesteps (#15021).  This can affect publish event timing, add spurious data to logs, etc.

The idea fix would be to have the binding for `AdvanceTo` automatically notice interrupts.  Until we have that, something like the above work-around is probably the simplest answer.

jwnimmer-tri, 2023-09-01:
I'll bet this is the right answer ...

https://pybind11.readthedocs.io/en/stable/faq.html#how-can-i-properly-handle-ctrl-c-in-long-running-functions

... along with some way to temporarily use it as a `set_monitor` on the Simulator.

PR #20130 (2023-09-11): [py systems] Simulator.AdvanceTo is interruptible by default
Closes #20053.

(Aside: This will probably make a bunch of Anzu simulations happier in CI.  As it stands, IIRC out CI has trouble killing simulator jobs and tends to time them out rather then interrupt them.)

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20130)
closed 2023-09-11 by rpoyner-tri