RobotLocomotion/drake #20054: [HardwareSim] `bazel run //examples/hardware_sim:demo_py` currently spews deprecation warnings

RussTedrake, 2023-08-24:


jwnimmer-tri, 2023-08-24:
If you have some sample output handy, could you post it?

Ah. Maybe it's these need to drop their extra submodule:
```
from pydrake.manipulation.kuka_iiwa import IiwaDriver
from pydrake.manipulation.schunk_wsg import SchunkWsgDriver
from pydrake.manipulation.util import ZeroForceDriver
```

RussTedrake, 2023-08-24:
```
/Users/russt/drake/examples/hardware_sim/hardware_sim.py:30: DrakeDeprecationWarning: Please import from the pydrake.manipulation module directly, instead of the deprecated pydrake.manipulation.kuka_iiwa submodule. The deprecated code will be removed from Drake on or after 2023-11-01.
  from pydrake.manipulation.kuka_iiwa import IiwaDriver
/Users/russt/drake/examples/hardware_sim/hardware_sim.py:31: DrakeDeprecationWarning: Please import from the pydrake.manipulation module directly, instead of the deprecated pydrake.manipulation.schunk_wsg submodule. The deprecated code will be removed from Drake on or after 2023-11-01.
  from pydrake.manipulation.schunk_wsg import SchunkWsgDriver
/Users/russt/drake/examples/hardware_sim/hardware_sim.py:32: DrakeDeprecationWarning: Please import from the pydrake.manipulation module directly, instead of the deprecated pydrake.manipulation.util submodule. The deprecated code will be removed from Drake on or after 2023-11-01.
  from pydrake.manipulation.util import ZeroForceDriver
```

PR #20077 (2023-08-28): [hardware_sim] Fix deprecation warnings
Closes #20054.
Amends #19869.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20077)
closed 2023-08-28 by rpoyner-tri