RobotLocomotion/drake #20061: [HardwareSim] BuildIiwaControl doesn't use lcm channel_suffix.

RussTedrake, 2023-08-24:
### What happened?

I believe that cannot use the `ApplyDriverConfig` with `IiwaDriver` for bimanual setups.  It's true that DrakeLcmParams have a `channel_suffix`, but that is not currently used in `BuildIiwaControl`
https://github.com/RobotLocomotion/drake/blob/339d852e21436db594e0faab9fe2d76fc3dd76fd/manipulation/kuka_iiwa/build_iiwa_control.cc#L90

jwnimmer-tri, 2023-08-25:
The suffix is associated with the `DrakeLcm` object itself.  Every call to `Subscribe` or `Publish` with a channel name will automatically tack on the suffix, under the hood, without any changes to the species' individual driver code.

Something like this, perhaps:

```yaml
Demo:
  directives:
  - ...
  lcm_buses:
    left_lcm:
      channel_suffix: _LEFT
    right_lcm:
      channel_suffix: _RIGHT
  model_drivers:
   left_iiwa: !IiwaDriver
      lcm_bus: left_lcm
   right_iiwa: !IiwaDriver
      lcm_bus: right_lcm
```

RussTedrake, 2023-08-25:
That resolved it.  Thanks!

closed 2023-08-25 by RussTedrake