RobotLocomotion/drake #20076: Functions to convert between SymPy and pydrake.symbolic.Expression

jwnimmer-tri, 2023-08-26:
**Is your feature request related to a problem? Please describe.**

Drake uses a custom [symbolic expression](https://drake.mit.edu/pydrake/pydrake.symbolic.html#pydrake.symbolic.Expression) module to support inter-operation between our C++ and Python code.  However, for users who need more complicated symbolic manipulation in Python, sometimes the features of our Expression class fall short (e.g., #8515, etc.).

[SymPy](https://www.sympy.org/en/index.html) is a very nice symbolic math package which can often provide the necessary additional features.

To help users leverage both libraries to their advantage, Drake should offer functions to convert from our own custom symbolic expressions into SymPy, and vice versa.

**Describe the solution you'd like**

An optional module under `pydrake` that (when imported) offers some kind of Python API to copy a Drake Expression (or matrix of expressions) into SymPy object(s), and vice versa.  This module would require the user to bring their own `SymPy`.  Drake wheels would not depend on SymPy.

**Describe alternatives you've considered**

Users could write their own converters using [Unapply](https://drake.mit.edu/pydrake/pydrake.symbolic.html#pydrake.symbolic.Expression.Unapply) but its probably subtle and boring.  Providing a first-party solution would be easier and more robust.

drewhamiltonasdf, 2023-08-26:
@jwnimmer-tri This is a great, great feature. I was considering writing something myself when you posted this by just parsing the strings from to_string() using RegEx or something along those lines. I'd be happy to give it a shot, but maybe you have a cleaner solution?

jwnimmer-tri, 2023-08-26:
For Expression => SymPy, the [Unapply](https://drake.mit.edu/pydrake/pydrake.symbolic.html#pydrake.symbolic.Expression.Unapply) will be the way to go.  We would call `e.get_kind()` to figure out which operation, and then call Unapply to get the operands, which we can pass to the analogous SymPy operator.

The only other trick will be some kind of memo dictionary, to convert between a Drake variable and a SymPy symbol.  We need to be careful to use the variable's identity, not name, in that mapping.

drewhamiltonasdf, 2023-08-26:
@jwnimmer-tri  Ok. I _think_ I get the jist of your idea. I would need to recurse through an expression, Sympy-ifying, generating smaller and smaller sub-expressions until nothing is left. Is that close?

Is this something I could do in Python by checking the type of _ctor_, or is this something best coded up in C++.

It also occurred to me that I'm being a little presumptuous thinking that I could help. I'm in a bit of a want-this-right-now mindset at the moment ;) I spent a bunch of time setting up the infrastructure to use Drake for a task and there's no backing out now.

I may just take a stab at it in Python this evening either way...

jwnimmer-tri, 2023-08-27:
> Ok. I think I get the gist of your idea. I would need to recurse through an expression, Sympy-ifying, generating smaller and smaller sub-expressions until nothing is left. Is that close?

Sounds right.

> Is this something I could do in Python by checking the type of ctor, or is this something best coded up in C++.

The `ctor` is useful to generically re-apply the operator.  Since we don't want Drake's operator, it won't help much.  The better answer is to ask [e.get_kind()](https://drake.mit.edu/pydrake/pydrake.symbolic.html#pydrake.symbolic.Expression.get_kind) to return an `ExpressionKind` enum.  We would if-else on that enum to figure out which SymPy operator to call.

The rough idea looks like this:

```python
def drake_expression_to_sympy(e: Expression):
    kind = e.get_kind()
    ctor, args = e.Unapply()
    if kind == ExpressionKind.Constant:
        value, = args
        return value
    if kind == ExpressionKind.Var:
        variable, = args
        return drake_variable_to_sympy_symbol(variable)
    if kind == ExpressionKind.Add:
        return functools.reduce(operator.add, args)
    if kind == ExpressionKind.Mul:
        return functools.reduce(operator.mul, args)
    if kind == ExpressionKind.Div:
        return functools.reduce(operator.truediv, args)
    if kind == ExpressionKind.Cos:
        arg, = args
        return sympy.cos(arg)
    raise NotImplementedError(f"Cannot convert {kind} yet")

def drake_variable_to_sympy_symbol(v: Variable):
    v_id = v.get_id()
    # ... magic happens here ...
```

jwnimmer-tri, 2023-08-27:
For the SymPy => Drake Expression direction, probably calling [lambdify](https://docs.sympy.org/latest/tutorials/intro-tutorial/basic_operations.html#lambdify) and passing in `pydrake.symbolic` as the target module would suit.

drewhamiltonasdf, 2023-08-27:
@jwnimmer-tri  I have a Python version pretty much wrapped up, but I made the mistake of running it with the LaTeX pretty printing option turned on and testing it on my 18x18 inertia matrix. That chews up enough RAM that I actually can't open the notebook anymore. We may need to manually over-ride the pretty-printing options and warn users if the sizes are too large. Will post tomorrow, when I get around to copy-pasting into Pycharm.

I suppose I was plenty forewarned about trying to get explicit symbolic representations of large systems, but... the heart wants what the heart wants. ;)

That said, I do think new users like myself will find this really useful for getting our sea legs (and doing homework), with smaller systems.

drewhamiltonasdf, 2023-08-27:
Just got back to this this afternoon.

My first stab at sympy to pydrake.symbolic produced an expression that looked great, but GetVariables() returned something nasty like: <Variables "{x, x, x, x, y, x, y, x, x, x, x, y, x, x, x, x, x, x, x, x, x, y, x, y, x}"> and no guarantee that the Variable type was the same as the original ("Continuous, Integer, Binary, Boolean, Random_uniform, etc.)

I'm considering requiring that the sympy->drake converter take the GetVariables() output from the original expression, and return an expression uses the same variables from the original (presumably) immutable std::set<Variable>

Should I keep pursuing this? Should I post what I have? Am I overthinking it?

drewhamiltonasdf, 2023-08-31:
@jwnimmer-tri Would it be fair to say that I'm missing the mark by a large margin here? Looked back at the top and seeing you want no Sympy dependencies and realizing I'm probably pretty far off target...

jwnimmer-tri, 2023-08-31:
The code here would go into a separate pydrake (sub-)module, and it's OK for _that_ module to depend on SymPy.  So long as the `import sympy` is isolated to that sub-module, and nothing else in `pydrake` touches it, it's fine.

One question we need to solve is what to call that module.  Maybe `pydrake.symbolic.sympy`?  The user would do `from pydrake.symbolic.sympy import pydrake_to_sympy, sympy_to_pydrake` or the like.  Any votes?

The code is roughly in the right direction.  The thing we need to be very careful about is how Variable objects are converted, and I'm not sure that it's doing that soundly yet.

To get this feature into Drake, the real work will be solving the Variable question, and then writing a clear and comprehensive unittest program that validates all of the kinds of conversions we care about.

At some point when I have free time, I was going to open a WIP PR with the skeleton of the new module and some test cases to get the ball rolling.  Then people who want this feature could push improvements to that branch.

jwnimmer-tri, 2023-09-02:
See #20117 for a rough outline.

jwnimmer-tri, 2023-09-02:
The code in #20117 is somewhat OK now, and feedback is welcome.  There are test wheel linked from the PR overview.

drewhamiltonasdf, 2023-09-03:
> The code in #20117 is somewhat OK now, and feedback is welcome. There are test wheel linked from the PR overview.

Great. I'll take a look. I am currently on vacation, but looking forward to trying it out!

jwnimmer-tri, 2023-10-10:
Any feedback yet, from @drewhamiltonasdf or anyone else?

drewhamiltonasdf, 2023-10-10:
Sorry sorry. I've been putting off figuring out how to install the branch. I'm working on O(n^2) symbolic matrix inverse for manipulator form equations actually as we speak. I'm not 100% certain I know how to get it running. Do I pull your fork and build from source 🤔. Bit of a noob...

jwnimmer-tri, 2023-10-11:
Ah, wow.  I didn't get an email with the prior message was retcon'd from a "How do I run it?" to "Here's some feedback".

Anyway, I launched wheel builds and those URLs are posted here now: https://github.com/RobotLocomotion/drake/pull/20117#issuecomment-1757780242.  They can be used in a virtual environment via `pip install https://...`.

I look at the other feedback shortly.

jwnimmer-tri, 2023-10-11:
> The conversions need to allow for integer inputs as well ... [and] ... make powers look nicer ...

Thanks for the feedback!

Yep, that all sounds fair.  Let's plan that I'll work to merge this PR as-is (with a TODO to disclaim that integers are not working yet), and then you can open a new PR with your improvements.

PR #20117 (2023-10-19): [pydrake] Add conversion to/from SymPy
Closes #20076.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20117)
closed 2023-10-19 by jwnimmer-tri