RobotLocomotion/drake #20131: Segmentation Fault in KinematicTrajectoryOptimization when Using MinimumDistanceConstraint

cohnt, 2023-09-05:
### What happened?

When constructing a `MinimumDistanceConstraint` and applying it at many points along a trajectory (for example, in `KinematicTrajectoryOptimization`), the way in which the constraint (and its corresponding context) are constructed can lead to errors or segmentation faults for unclear reason. I've done my best to make a [standalone example](https://gist.github.com/cohnt/97fe842675dad3b1c4c6f97bf40ed5fc), using code from [manipulation](https://github.com/russtedrake/manipulation). I've highlighted the relevant changes below -- it seems that re-creating the context or the constraint is causing the error.

[Original code](https://gist.github.com/cohnt/97fe842675dad3b1c4c6f97bf40ed5fc#file-original-py)
```
    # collision constraints
    collision_constraint = MinimumDistanceConstraint(
        plant, 0.001, plant_context, None, 0.01
    )
    evaluate_at_s = np.linspace(0, 1, 25)
    for s in evaluate_at_s:
        trajopt.AddPathPositionConstraint(collision_constraint, s)

```

[Modification to induce an error](https://gist.github.com/cohnt/97fe842675dad3b1c4c6f97bf40ed5fc#file-error-py)
```
    # collision constraints
    evaluate_at_s = np.linspace(0, 1, 25)
    for s in evaluate_at_s:
        context = diagram.CreateDefaultContext()
        plant_context = plant.GetMyContextFromRoot(context)
        collision_constraint = MinimumDistanceConstraint(
            plant, 0.001, plant_context, None, 0.01
        )
        trajopt.AddPathPositionConstraint(collision_constraint, s)
```

[Modification to induce a segmentation fault](https://gist.github.com/cohnt/97fe842675dad3b1c4c6f97bf40ed5fc#file-segfault-py)
```
    # collision constraints
    evaluate_at_s = np.linspace(0, 1, 2)
    for s in evaluate_at_s:
        context = diagram.CreateDefaultContext()
        plant_context = plant.GetMyContextFromRoot(context)
        collision_constraint = MinimumDistanceConstraint(
            plant, 0.001, plant_context, None, 0.01
        )
        trajopt.AddPathPositionConstraint(collision_constraint, s)
```

### What operating system are you using?

Unknown / Other, Ubuntu 22.04

### What installation option are you using?

compiled from source code using CMake

hongkai-dai, 2023-09-05:
Could you try one more thing here? To create the contexts outside of the for loop
```py
    # collision constraints
    evaluate_at_s = np.linspace(0, 1, 2)
    contexts = [None] * 2
    plant_contexts = [None] * 2
    for i in len(evaluate_at_s):
        contexts[i] = diagram.CreateDefaultContext()
        plant_contexts[i] = plant.GetMyContextFromRoot(contexts[i])
        collision_constraint = MinimumDistanceConstraint(
            plant, 0.001, plant_contexts[i], None, 0.01
        )
        trajopt.AddPathPositionConstraint(collision_constraint, evaluate_at_s[i])
```
Do yo still see the segmentation fault?

In C++, if I create a `context` and `plant_context` within the loop, then the lifetime of that plant_context is just within the for loop. As soon as the for loop finishes, the `context` and `plant_context` are destroyed. I would think pybind does something similar? But it should be fine if the contexts are created and stored outside of the for loop.

cohnt, 2023-09-05:
Your change works -- no segmentation fault.

jwnimmer-tri, 2023-09-05:
Ideally, the `MinimumDistanceConstraint` argument `plant_context` should be annotated to "keep alive" whatever is passed in.

In fact it already seems to do that:

```c++
// Keep alive, reference: `self` keeps `plant_context` alive.
py::keep_alive<1, 4>()
```

Maybe `GetMyContextFromRoot` is missing an annotation?  Nope, it seems to have it as well:

```c++
// Keep alive, ownership: `return` keeps `Context` alive.
py::keep_alive<0, 2>()
```

And `AddPathPositionConstraint` uses `shared_ptr` so should ref-count automatically correctly.

Someone will need to dig in and figure out where the reference counting goes awry.  I don't immediately see any problem with the bindings.

hongkai-dai, 2023-09-05:
I am going to delegate this issue to other pybind gurus. Anybody volunteer?

jwnimmer-tri, 2023-09-06:
Per https://drake.mit.edu/issues.html we can assign to the component lead, +@EricCousineau-TRI.

EricCousineau-TRI, 2023-10-17:
Ran into possibly related segfault in Anzu, testing things out.

I can reproduce the failure locally using your code (thanks!)
https://github.com/RobotLocomotion/drake/tree/3115855343b2e717e008b3e5a3e01ecdc0334e80/tmp
(this includes pybind11 upgrade, #20362, but that doesn't seem to change anything as far as this issue is concerned)

From adding trace (per [debugging tips](https://drake.mit.edu/python_bindings.html#debugging-with-the-python-bindings)), I see the segfault happen as so:

```sh
$ bazel-bin/tmp/repro
...
repro.py(169):         traj = BsplineTrajectory(
repro.py(172):         values = traj.vector_values(np.linspace(0, 1, 50))
repro.py(173):         print(values)
...
repro.py(174):         meshcat.SetLine("positions_path", values)
Segmentation fault (core dumped)

```

EricCousineau-TRI, 2023-10-17:
Looks like the `shared_ptr<>` object is either getting sliced or lost; see #20380

EricCousineau-TRI, 2023-10-17:
Confirmed, posted a new issue w/ writeup: #20383

jwnimmer-tri, 2024-11-02:
Closing as duplicate of #20383.

cohnt, 2025-03-19:
@rpoyner-tri given that #20383 has been closed (and as far as I can tell, so have all other downstream issues), but this still segfaults on master, perhaps we should reopen this PR? Otherwise, I don't see how the issue is going to get tracked.

rpoyner-tri, 2025-03-21:
Per slack discussion with @cohnt, reopening. I will take a look.

rpoyner-tri, 2025-03-21:
I resurrrected @EricCousineau-TRI 's #20380 . The test shown there still fails vs. master.

PR #22795 (2025-03-27): [pydrake] Fix costs and constraints memory hazards
Closes #20131.

---
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/22795)
closed 2025-03-27 by SeanCurtis-TRI