RobotLocomotion/drake #20133: pointcloud_voxelization_test appears flakey on macOS

svenevs, 2023-09-05:
The test `//tools/workspace/voxelized_geometry_tools:pointcloud_voxelization_test` seems to be failing on macOS (arm64 _and_ x86_64).  CI has been happy with them for quite some time, but locally this test fails for me on any clean build.

https://github.com/RobotLocomotion/drake/commit/72c24dc29fd804017ddbf3891bfe4ba3339fe4b7

The bisect was a bit strange, there were a couple commits that if I re-ran it I could get it to fail or succeed (seemed arbitrary...).  Factoring out of https://github.com/RobotLocomotion/drake/pull/19945#pullrequestreview-1612040605

CC @jwnimmer-tri for delegation / prioritization please.

jwnimmer-tri, 2023-09-05:
After the CI load calms down later, my plan is to fire off a few PR test builds to see if it reproduces in CI (by setting `tags` that force it to be uncached.)

jwnimmer-tri, 2023-09-06:
\CC @calderpg-tri FYI.  At the moment, my guess is this is a CI infra problem.  In case I get any evidence otherwise, I might ping you for more help.

jwnimmer-tri, 2023-09-07:
Forcing the test to re-run on arm64 [passed without any problem](https://drake-jenkins.csail.mit.edu/job/mac-arm-ventura-clang-bazel-experimental-release/2155/console).   (I nerfed a bunch of unrelated stuff, so the overall job failed for other reasons, but that `PASSED` line for this test is there, green, and uncached.)

Also remember that macOS x86_64 doesn't have any remote caching, so was already running on every single [mac-x86-monterey-clang-bazel-continuous-release](https://drake-jenkins.csail.mit.edu/view/Continuous%20Production/job/mac-x86-monterey-clang-bazel-continuous-release/) since forever.

I think this problem is only that "the tests fail on @svenevs for some reason that is not captured by CI".

svenevs, 2023-09-07:
> "the tests fail on @svenevs for some reason that is not captured by CI".

Sounds right to me.  I'm happy to just say "CI is happy" and reopen this if it resurfaces.

calderpg-tri, 2023-09-07:
I have been able to reproduce flaky behavior in the test on my M2 machine, for both serial and parallel kernel dispatch. I haven't looked at the intermediate tracking and filter grids yet, but the nature of the test failures makes me suspect the issue has to do with kernel dispatch/sequencing - an explanation of the failures would be that the filter phase is somehow running before the raycasting phase has completed.

calderpg-tri, 2023-09-10:
From more testing, I strongly suspect that something with the macOS implementation of `clEnqueueBuffer` is just *broken*. Looking at the pre-raycasting and post-raycasting tracking grid data, it's full of repeating garbage values that change from run to run, but generally using much of the range of an `int32_t`. If I replace the current [allocate + zero](https://github.com/ToyotaResearchInstitute/voxelized_geometry_tools/blob/master/src/voxelized_geometry_tools/opencl_voxelization_helpers.cc#L382-L407) with an allocate-as-copy buffer using zeroed host memory, everything works.

PR #20163 (2023-09-10): Bump VGT with workaround for macOS OpenCL and fixes for Intel OpenCL
Changes in VGT:
- Workaround for what appears to be a broken `clEnqueueFillBuffer` on macOS (resolves #20133)
- Fixes for OpenCL kernel double->float conversions, which are now warnings using the latest Intel OpenCL (specifically necessary to support ADL-N and ARC GPUs)
- Updates to buffer memory flags and kernel build flags

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20163)
closed 2023-09-10 by jwnimmer-tri