RobotLocomotion/drake #20149: Verify pip index url lifecycle rules

svenevs, 2023-09-07:
As part of https://github.com/RobotLocomotion/drake-ci/pull/241 I added a lifecycle rule that I _believe_ will start deleting non-current versions after one day.  I want to monitor this over a few days and make sure (the PR has not even merged yet).  The way to create a rule to only delete noncurrent things was a little strange, you can't use the value `0` for either field, have to do `1` day and `1` max noncurrent versions.

[link to the lifecycle rule in question](https://s3.console.aws.amazon.com/s3/management/drake-packages/lifecycle/view?region=us-east-1&id=Rule+for+whl/nightly+pip+indices)

jwnimmer-tri, 2023-09-07:
Hmm.  However -- in case a nightly build fails, we should _not_ delete the prior day's html index.  A deletion period of 7 days or so would be better.  (Leaving it at 1 day during testing is OK, though.)

jwnimmer-tri, 2023-09-07:
> ... that "current" versions never get deleted.

Perfect!

svenevs, 2023-09-14:
Plan is to recheck things on Tuesday 9/19 to make sure things are deleting as we expect.

svenevs, 2023-09-15:
The version from September 12th is now gone.  I'll probably be around at some point this weekend and if I remember to check resolve this.  I believe what we're seeing is that the lifecycle rule is correct, but we don't control when it actually gets garbage collected.  If I understand how it works, there are "delete markers" pushed out into a queue, and those get processed at some indeterminate (and likely highly optimized) interval.

svenevs, 2023-09-19:
September 17th is the latest non-current version, 7 versions seem to be tracked at a time, we're good here.

closed 2023-09-19 by svenevs