RobotLocomotion/drake #20150: GetInfeasibleConstraintNames does not work with custom constraint function in pydrake

yu-fz, 2023-09-07:
### What happened?

MathematicalProgramResult.GetInfeasibleConstraintNames seems to be broken for Mathematical Programs that uses custom defined constraint functions.

### Repro
```
from pydrake.solvers import (
    Solve)

from pydrake.all import(
    MathematicalProgram,)

from pydrake.autodiffutils import (
    AutoDiffXd,
    ExtractValue, 
    InitializeAutoDiff)
import numpy as np

prog = MathematicalProgram()
# add decision variables
x = prog.NewContinuousVariables(2, "x")
# add quadratic cost
prog.AddCost(x[0]*x[0])
# Add custom constraint function
def bad_constraint_fn(x):
    if isinstance(x[0], AutoDiffXd):
        a = ExtractValue(x)
    else:
        a = x
    return InitializeAutoDiff(value=np.array([a[0]*a[1]]),
                              gradient=-np.array([10]))

# add custom infeasible constraint, this breaks GetInfeasibleConstraintNames
prog.AddConstraint(bad_constraint_fn, 
                   lb=[-1], 
                   ub=[-1], 
                   vars=x).evaluator().set_description("bad_constraint_1")
# This is OK.
prog.AddLinearEqualityConstraint(Aeq=np.array([[1, 2],[2,1]]),
                                 beq=np.array([-np.inf,np.inf]),
                                 vars=x).evaluator().set_description("bad_constraint_2")
                                 
result = Solve(prog)
print(result.is_success())
if not result.is_success():
    # this line fails
    constraints = result.GetInfeasibleConstraintNames(prog)
    for c in constraints:
        print(c)
```
        

### Version

Drake 1.20.0

### What operating system are you using?

Other

### What installation option are you using?

pip install drake

### Relevant log output

```shell
File "/Users/fangzhou/dev/SRB_Trajopt/srb_trajopt/repro_drake_infeasible_constraints.py", line 41, in <module>
    constraints = result.GetInfeasibleConstraintNames(prog)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/fangzhou/miniconda3/envs/srb_trajopt_env/lib/python3.11/site-packages/pydrake/solvers/_extra.py", line 35, in _check_array_type
    raise RuntimeError(
RuntimeError: PyFunctionConstraint: Output must be of scalar type float. Got AutoDiffXd instead.
```

jwnimmer-tri, 2023-09-07:
The constraint function is buggy.  When `isinstance(x[0], float)`, the return type should be `float` not `AutoDiffXd`.

Drake should try to improve its error message in this case, though.

jwnimmer-tri, 2023-09-08:
On further thought, I wonder if it would be better to silently discard the derivatives and allow the function to always return a gradient, even when one was not necessary?

WDYT @EricCousineau-TRI @hongkai-dai ?

EricCousineau-TRI, 2023-09-08:
For the Python side of things, that'd work for me!

hongkai-dai, 2023-09-08:
> On further thought, I wonder if it would be better to silently discard the derivatives and allow the function to always return a gradient, even when one was not necessary

Sorry I am not sure if I understand what this means.

Do you mean that the python function (`bad_constraint_fn` in this example) always takes in a numpy of AutoDiffXd and returns a numpy of AutoDiffXd, and in `Constraint::Eval(const Eigen::Ref<const Eigen::VectorXd>& x, Eigen::VectorXd* y)` we just do
```
PyFunctionConstraint::Eval(const Eigen::Ref<const Eigen::VectorXd>& x, Eigen::VectorXd* y) {
  x_ad = x.cast<AutoDiffXd>();
  AutoDiffVecXd y_ad;
  bad_constraint_fun(x_ad, y_ad);
  *y = math::ExtractValue(y_ad);
}
```
? What if I do want my python function to evaluate both numpy array of double and numpy array of AutoDiffXd?

EricCousineau-TRI, 2023-09-08:
I took it to mean we special case where a user implements a function that only returns AutoDiffXd. On the C++ side, if the overload is intended for `double`, and the user's Python function returns `AutoDiffXd`, use `.cast<double>()` on the vector; this will incur a small performance penalty, but will prevent a user from stumbling across a situation like this.

jwnimmer-tri, 2023-09-08:
For the input argument, I'm not proposing any change (i.e., we give the user an `ndarray` of `float` when we don't want the gradient, and `ndarray` of `AutoDiffXd` when we do want the gradient).

The proposed change would be to the return value -- if we pass the evaluator a float we allow a return value of either float or autodiff.  (If we pass the evaluator an autodiff, we still require an autodiff return.)

hongkai-dai, 2023-09-08:
Got it, thanks for the explanation.

yu-fz, 2023-09-12:
I never understood why the constraint functions can have either `float` or `AutoDiff` input arguments passed to it. Does SNOPT decide when it does not require gradients of the constraint function?

hongkai-dai, 2023-09-12:
> I never understood why the constraint functions can have either float or AutoDiff input arguments passed to it. Does SNOPT decide when it does not require gradients of the constraint function?

SNOPT will always need to compute the gradient. Hence when the constraint is evaluated inside SNOPT, we will always use the constraint function with AutoDiff argument.

But when you call `GetInfeasibleConstraintNames`, it doesn't need the constraint gradient, only the constraint value, hence `GetInfeasibleConstraintNames` uses the constraint function with float input arguments.

jwnimmer-tri, 2023-09-12:
See #13234 for the prior work on error messages.  We're already creating this message ourselves, it's not `pybind11`:

https://github.com/RobotLocomotion/drake/blob/c482ed7f6222958eb1f2904db9bbbd22e6eedf9f/bindings/pydrake/solvers/_extra.py#L33-L37

jwnimmer-tri, 2023-09-13:
It was a bit tricky to automatically fix the return type, so I abandoned that approach for now.  In #20178, I've just improved the error message:

`TypeError("When PyFunctionConstraint is called with an array of type float the return value must be the same type, not AutoDiffXd")`

PR #20178 (2023-09-21): [py solvers] Improve error message for evaluator return type mistakes
Closes #20150.

- - -
This change is [[image]](https://reviewable.io/reviews/RobotLocomotion/drake/20178)
closed 2023-09-21 by EricCousineau-TRI