RobotLocomotion/drake #20152: Roadmap for making glTF a first-class citizen

SeanCurtis-TRI, 2023-09-07:
As of September 2023, glTF has gained a significant foothold in Drake. It is generally superior to obj for representing objects for *visual* tasks (the illustration and perception role). We want to push it further such that we can declare that people should use glTF without caveat and reservation. This represents a list of tasks toward that goal.

The list is fluid. Items may be introduced, priorities may be changed. What matters most is that the list includes the things we care about (even if it is for far off in the future). As appropriate, each item should map to an issue to facilitate discussion (not all issues truly require discussion).

## Minimum viable product
- [ ] Doxygen module describing Drake's support for geometry files (kept up to date as subsequent tasks get completed).
- [ ] Minimum `RenderEngineGl` support (convert PBR materials to simply Phong materials currently supported.
- [ ] RenderEngineVtk features
    - [ ] Environment map (HDR)
    - [ ] Shadows
- [ ] Meshcat
    - [ ] HDR environment maps
- [ ] Update `drake_models` to use glTF files
    - [ ] Also address #19935
- [ ] #20151

## Future work
- [ ] Support .gltf with external references.
- [ ] Support .glb files.
- [ ] Use glTF in proximity role.
- [ ] `RenderEngineGl` features:
    - [ ] PBR shader.
    - [ ] Shadows
    - [ ] Environment map (HDR)
- [ ] #18743

SeanCurtis-TRI, 2023-09-07:
cc @jwnimmer-tri

jwnimmer-tri, 2023-09-10:
Thanks for this!

Our [preference](https://drake.mit.edu/issues.html#component) is to have one "component" label per issue, when we can.  Could we consider `component: geometry general` and `Epic` labels for this issue, and then use the more specific components on the sub-issues?

SeanCurtis-TRI, 2023-09-12:
As I re-read the label guidance, I note:

> Our preference is to have exactly one `component` per issue, but we allow multiple in case several components are equally relevant.

I'll ruminate on that while I decide if I restore the multiple labels. And I think it's easy to argue that the `Epic` label does a lot for justifying multiple components.

SeanCurtis-TRI, 2026-01-05:
The shopping list partitions MVP vs future. The "VTK exports glTF files rotated" thing would be nice to fix. However, the one application that depends on it is accounting for the silliness (which, of course, heightens the cost of fixing the defect).

We can roll that one item into its own issue or leave this open to track it. Either way, we end up with the same number of issues.