RobotLocomotion/drake #20153: RenderEngineGltfClient misses extensions on input glTF files

SeanCurtis-TRI, 2023-09-07:
### What happened?

`RenderEngineGltfClient` generates glTF data and then merges input glTF data into it. The result is what is passed to the server. Currently, it is explicitly omitting several groups of glTF data (e.g., animation and morph targets). However, it is also skipping extensions. This is incorrect.

To maintain maximum compatibility between input glTF and output glTF, it should include the two lists: "extensionsUsed" and "extensionsRequired".

It *appears* that the exercise of those extensions within various entities (e.g., materials) seems to be working. But they must also be declared.

jwnimmer-tri, 2024-02-13:
On first glance, Blender supports [KHR_draco_mesh_compression](https://github.com/KhronosGroup/glTF/blob/main/extensions/2.0/Khronos/KHR_draco_mesh_compression/README.md]) so preserving it at least in `extentionsUsed` would a very nice.

(And of course, we might as well preserve `extensionsRequired` while we're at it, since it's essentially the same fix.)

jwnimmer-tri, 2024-02-13:
Actually, if I'm reading correctly I think #20184 fixed this already.  @SeanCurtis-TRI please re-open if I'm mistaken.

closed 2024-02-13 by jwnimmer-tri

SeanCurtis-TRI, 2024-02-13:
That was certainly the intention of #20184. It's possible that it didn't do what it aspired to do. Time will tell. But, yes, that should've closed this issue.