RobotLocomotion/drake #20165: Port Panda drivers for hardware_sim (PandaArmDriver and PandaHandDriver) from Anzu?

RussTedrake, 2023-09-10:
Related to #20055 (and it makes sense to resolve that one first).

Many users of HardwareStation want to run on panda.  Can we port those "drivers" to Drake?

jwnimmer-tri, 2023-09-10:
For the arm ...

This would be for simulation only, is that right?  There is no `https://github.com/RobotLocomotion/drake-panda-driver` repository with any actual hardware I/O.  In that case, would we rather have a plain `InverseDynamicsDriver` that is not specific to any species?  (With configurable gains, filtering, and control mode.)  That would let users simulate many kinds of arms, without editing Drake for each new kind of arm.

For the hand ...

I'm not sure about copying it from Anzu to Drake as-is.  There is a comment atop the file "THIS IS AN INTERIM HACK AND SHOULD ALL BE REPLACED WITH A PROPER PANDA HAND CONTROLLER!" which gives me pause.  Someone would need to go look at the code to see what (if any) parts of it are competent enough for Drake.

jwnimmer-tri, 2024-11-01:
> In that case, would we rather have a plain InverseDynamicsDriver that is not specific to any species?

Hearing no reply, I'm going to decide "yes" and therefore adding **panda** specific driver code into drake is not planned.  If we want to add some generic implicit_pd-based robot controller in Drake that would work with any robot species, that's probably a good idea, but should be opened as a different issue if so.

Closing as "not planned".

closed 2024-11-01 by jwnimmer-tri